smartavionics/Cura #3: V20190311 - Skin Edge Support

amagr0, 2019-03-11:
Hello @smartavionics,

I'm really curious about your last feature - "Skin Edge Support", but I'm not found it.
Could you please confirm that this feature is already implemented on "20190311" windows version?
(where can I find it?)

Tks

smartavionics, 2019-03-11:
Hi @amagr0, it will be available later today, I'm just making a few changes to it.

smartavionics, 2019-03-11:
What it does is put an infill wall underneath the edge of the skin regions above. Here's an example...

Here's the infill with the added wall(s)...

[image]

Here's the next layer, you can see the skin is very well supported...

[image]

And here is the object printed. This is actually using just 1 wall and 1 skin layer over 10% infill. This object weighs very little and is very stiff considering the small amount of filament used....

[image]

Without the infill wall below the skin edge you would expect to see holes in the skin where it meets the wall.

amagr0, 2019-03-11:
Really clever, as usually! ;)
Is already available?

smartavionics, 2019-03-11:
>  Is already available?

Not yet, I will be uploading a Windows release to dropbox very soon.