smartavionics/Cura #4: V20190311 - Inner wall lines trespassing the model

amagr0, 2019-03-11:
**Application Version** - CURA Master - V20190311
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

I'm getting this strange inner wall lines trespassing the model:

Layer height: 300 micron
[image]

Layer height: 200 micron
[image]

Layer height: 100 micron
[image]

[inner_lines.zip](https://github.com/smartavionics/Cura/files/2951904/inner_lines.zip)

smartavionics, 2019-03-11:
Thanks for the report, I can confirm the bug. I will investigate... The only workaround I think is to disable the walll gap filling.

smartavionics, 2019-03-11:
Hi @amagr0 , I think that I have fixed this problem. The dropbox release 20190311 has been updated so please try it when you can. Thanks.

amagr0, 2019-03-11:
Yes, it seems to be solved.

Layer height: 300 micron
[image]

Layer height: 200 micron
[image]

Layer height: 100 micron
[image]

closed 2019-03-11 by amagr0