smartavionics/Cura #5: Feature request - Minimum extrusion length

amagr0, 2019-03-11:
Is there any way to not extrude line segments lower that a defined length? 

In the image below there is some really short inner wall segments, it will be great if we could define for example that all the segments lower than x mm will not be extruded.

[image]

smartavionics, 2019-03-11:
There is a cutoff for the size of the filled areas. Areas less than 0.25 mm^2 are ignored. It would be possible to make that value a setting.

amagr0, 2019-03-11:
That's related only for the infill, it will be nice to have a setting for all extruded segments (infill, walls, top, bottom, etc)
I don't think that feature is working has respect, I chose 10mm^2 and those little infill segments are there.

[image]

smartavionics, 2019-03-11:
Sorry, that cutoff value is hard coded into the wall gap fill code, it isn't the setting you mention above. I would need to add another setting to give you what you are requesting.

amagr0, 2019-03-11:
That setting will be very helpful.
A solution to avoid the top/bottom holes caused by the usage of this setting, could be on deactivate this feature on top surface and first layer.

smartavionics, 2019-03-12:
Hi @amagr0 , could you please send me the project file for your example above? Thanks.

amagr0, 2019-03-12:
[suporte_fpv_am.zip](https://github.com/smartavionics/Cura/files/2957319/suporte_fpv_am.zip)

smartavionics, 2019-03-12:
There is also an existing setting called Filter Out Tiny Gaps - seems to do pretty much what you want here.

amagr0, 2019-03-12:
Yes, it seems to work well in this situation, but would be nice to take control of this parameter, and also an option to deactivate it on the last and first layers to avoid open little holes.

smartavionics, 2019-03-13:
From my README.md...

> 20190313 - New settings - Minimum Gap Area and Inner Layer Minimum Gap Area set the min size of areas that will be filled when Filter Out Tiny Gaps is enabled.

amagr0, 2019-03-13:
I tryed to change the values of the "Minimum Gap Area and Inner Layer Minimum Gap Area" and it seems that is not working.

In the image bellow you can see that is alredy on 10mm^2 and those little extrusions still there. 

[image]

[min_gap_area.zip](https://github.com/smartavionics/Cura/files/2962077/min_gap_area.zip)

smartavionics, 2019-03-13:
That's because those areas are infill and not the gap between walls. There is a setting called Minimum Infill Area that was originally intended to replace areas of infill that are smaller than the threshold with skin. I don't know if it still works because some time back it was behaving as if someone had broken it. Anyway, see if setting a value there helps.

amagr0, 2019-03-13:
You're right, but it seems that it continue broken.
Maybe I should create an issue on the Ultimaker/Cura github page.

[image]

smartavionics, 2019-03-13:
Remember, that skin only occurs at the top/bottom so infill that is not on a top/bottom layer will not get replaced if it is smaller than Minimum Infill Area.

amagr0, 2019-03-13:
So, there is no way to not extrude segments on the infill smaller that a specific area. 
I think this could be a challenge for your skills. ;)

smartavionics, 2019-03-13:
>  I think this could be a challenge for your skills. ;)

It's no challenge, so I am not interested :).

It already removes infill regions whose area is < 0.16mm^2 so those regions in your print must be larger than that.

amagr0, 2019-03-13:
It's hard-coded in Cura to remove infill regions less than 0.16mm^2, is that?

The problem is whem you start to use diferent nozzle diameters, (0,6...1.0mm - or even bigger) those areas stop making sense, I think the best way to resolve this is to work not em mm^2 (area) but in mm (line segment) (I don´t know how complex could be to implement this).

smartavionics, 2019-03-13:
It's a trivial change.

Right now the minimum area is 0.4 x 0.4 mm. Are they assuming that's approx the line width? In other parts of the code they use 2 x line_width^2 as the area threshold. I could use that instead?

amagr0, 2019-03-13:
If they are assuming a fix value like, 0,4x0,4, seams they are considerating an aproximation of the a 0,4 nozzle diameter, but it should be depend on the nozzle select by the user. like the "2 x line_width^2", and not on a fix value. 
It would be nice for the user to take control of this parameter.

smartavionics, 2019-03-13:
>  It would be nice for the user to take control of this parameter.

No, there's already far too many settings as it is. I am happy to make the value be scaled from some existing value (like infill line width) but I am not going to add another setting for this.

amagr0, 2019-03-13:
An example where this little extrusions could fail an entire print is with ceramic prints, where the nozzles are bigger that normally are. 
In general I'm just try to find a way to simplify the gcode trajectories avoiding all the unnecessary extrusions, in order to get more successfully prints.
Of course that I would like to see more autonomous Slicers (maybe on day with AI) to reduce all the manual configuration by the user, but unnfortunatly and even with all your amazing work, we stilll with Slicers 1.0.

smartavionics, 2019-03-14:
I have created a new release that uses the infill line size to scale the minimum infill area. Please see if it works any better for you.

amagr0, 2019-03-22:
Hello @smartavionics,

I'm still struggling with this micro depositions that are causing extrusion problems.
In the images bellow you can see some big travel moves in order to only extrude really small inner wall segments.

[image]

[image]

[image]

Should this last implemented feature "Inner Layer Minimum Gap Area" had solve this?

___

In this generated gcode there's something strange, there is a fill line on top of an already printed inner wall.

[image]

[image]

Thanks

[inner_wall_fill.zip](https://github.com/smartavionics/Cura/files/2997045/inner_wall_fill.zip)

smartavionics, 2019-03-22:
Hello @amagr0 , reducing the wall line count to 2 will remove the little green wall blobs.

As for the wall fill on top of the green wall line, I think that is an original Ultimaker bug rather than a new addition by me but I will investigate it anyway.

smartavionics, 2019-03-22:
I can confirm the filled wall on top of the inner wall as a bug but I can't provide a solution yet. As a workaround, enable the overlap compensation for the inner walls and set the min wall flow to a non-zero value (i.e 10).

smartavionics, 2019-03-23:
Hi. I have worked out what is going wrong here and it's a fundamental problem with how the wall lines are derived. Right now, I don't have a code fix. It occurs in the special situation where you have a wall where the width of the wall in some places is an exact multiple of the wall line widths and also there are also other regions where the wall width is wider. One  workaround is to use a wall line width that is slightly different so in your example using 0.41 for the inner and outer wall line width fixes the problem as the long green line doesn't get printed then.