smartavionics/Cura #6: Sme cosmetic things to a great Fork

FlailAway, 2019-03-18:
Thanks for your Fork of this. You are turning this into a way better Slicer then the original.
3.6.0
Mint 19.1/Cinnamon

Pardon me for nit-picking, but this software deserves to look as good as it works. 
The splash screen shows "Cura mast" with the last part of "Master" cut off.
After slicing the the dialog with "Preview" and "Save to file" buttons has the "Preview" button half hidden.
After slicing, at the top of the screen, buttons "Prepare" and "Preview" seem at odds with the drop down for "View type." I would suggest having another button at the top for "XRay" and do away with the drop down. Or add "Prepare view" to the drop down.
The drop down is a little confusing as the "Color Scheme" is suddenly hidden when dropped down and "XRay" selected. This also drags the Profile box to the left. having stuff moving about on the screen is confusing.

It's OK for you as you know what's happening, but to the new user, I had to read all of the things again to see what just happened and where I was now at on the screen. 

Overall, these are minor things as the working of the Fork is the guts and that is great. But I am a retired programmer and have started, run and later, sold a couple of software businesses over the years so I have a good feel for what looks right. :)

Great job on the thin walls though.

smartavionics, 2019-03-18:
Ah, right. Sounds like all your comments are related to the UI. Sorry, I don't do any work on the UI apart from adding settings that I need. I only put any time into the backend (CuraEngine). I know there are many little niggles with the UI as people are moaning continuously about it on the Ultimaker forum and also on github. Personally, I don't care that the UI has problems as long as I can actually get the gcode I want.

So, I think I should bounce you back to the Ultimaker forum or the Ultimaker/Cura issues on github. Sorry to waste your time.

closed 2019-07-28 by smartavionics

JohnEdwa, 2019-07-31:
You can also give [FieldOfView and the Sidebar GUI plugin a go.](https://github.com/fieldOfView/Cura-SidebarGUIPlugin) You can also find it on the Cura marketplace.

smartavionics, 2019-07-31:
It just so happens that today's release does fix the too big "master" in the splash. It also features the upstream fix for the z hop fiasco in the 4.2 release.