smartavionics/Cura #8: Dual extruder, single heater, mixing nozzle gcode order causes failure

andymozza, 2019-05-24:
Application Version
4.0
Platform
Windows 10 and others
Printer
M3D Promega
Dual extruder stepper, single mixing nozzle, single heater
Steps to Reproduce
When slicing for dual tools, Cura turns un-used tools or switched tools to 0 or the standby temperature, however it does this 'after' the temperature for the new tool is set.
Because this is the same shared heater, then the tool starts heading towards 0.... filament gets stuck and doesn't extrude etc.
M104 S200
G1 F600 E-2
T2
M109 S210
M104 T0 S0 <-- Set T0 to o degrees which affects the T2 which has already been switched to
Actual Results
Filament will no longer extrude because it's been cooled too much as the tool is switched off.
Expected results
M104 to turn the tool off should be sent 'before' the temperature for the new tool is set.
very simple fix, just switch the order of the temp gcode.
M104 T0 S0 <-- Switch off un-used tool first, then work on the used tool
M104 S200
G1 F600 E-2
T2
M109 S210
Additional Information
I realise there are other issues related to this which have been rejected as you don't have these tools or heads etc, but I think this simple change can be visually validated and will not affect others, but be of benefit to these mixing nozzles with single shared heater.
Selection of gcode from switch between brim printed with extruder 0 and then main object printed with extruder 2 as an example.

andymozza, 2019-05-24:
Created to track feedback on the proof of concept build:
[https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AACTew8TBE1UhXY0Kxuw0-91a/Cura-mb-master-win64-20190524.exe?dl=0](https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AACTew8TBE1UhXY0Kxuw0-91a/Cura-mb-master-win64-20190524.exe?dl=0)

Main dropbox folder (earlier and later builds):
[https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AACTew8TBE1UhXY0Kxuw0-91a/Cura-mb-master-win64-20190524.exe?dl=0](https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0)

smartavionics, 2019-05-24:
The dropbox folder link is actually...

https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

andymozza, 2019-05-24:
> The dropbox folder link is actually...
> https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Apologies - pasting error :-) Corrected.

andymozza, 2019-05-28:
For those that asked for more information: Adds a setting on printer configuration to mark the extruders as having a shared heater:
[image]

This then stops any spurious tool changes and temperature changes on un-used tools which cause the print to fail.

Jdbye, 2019-05-29:
I just have to say, thank you so much for making this. This issue (I would even go so far as to call it a bug) has been driving me crazy trying to figure out why it happened. It only seemed to happen when I was printing certain things, so I mistakenly thought it might be a problem with the stl file causing incorrect gcode to be made. But looking at the gcode, it quickly became apparent that the issue was with Cura.
I'm disappointed at how unwilling the Cura team seem to be to do anything about it. It's such a simple change. Well, hopefully this will be merged into the official master branch.

It seems to be working just fine so far. Of course there might be some weird edge cases where it introduces new bugs that weren't there previously, but my Geeetech A20M is much more usable in Cura now.

smartavionics, 2019-05-29:
> I'm disappointed at how unwilling the Cura team seem to be to do anything about it. It's such a simple change. Well, hopefully this will be merged into the official master branch.

Come on, be reasonable. You can't expect them to put effort into providing a feature that only benefits people using 3rd party printers. Kudos to Ultimaker for making Cura available for use with non-Ultimaker products and for taking the time to integrate contributions that don't provide any benefit to Ultimaker.

Jdbye, 2019-05-29:
It might have nothing to do with this fork, but I can only seem to print a 75/25 mix of color now, which is the last virtual extruder in my list. I have it set up with left, right, 25/75, 50%, 75/25. It started right after I switched to the fork, but I changed some other settings as well, I can't see what could be causing it though. There's only a T0 or T1 once in the gcode file (depending on whether I'm printing with the left or right extruder)

Jdbye, 2019-05-29:
But that's the thing. Cura is not only designed to work with Ultimaker printers, and an issue like this essentially makes it almost useless with some printers. That's a pretty big deal and they should want to fix it so that people can use their software. Right? But they don't even seem interested.
If they said something like "make a pull request" that would be one thing but they just seem to have zero interest in more printers working well with their software. 
I guess I came off a bit harsh though. But after hearing all the people raving about how good Cura is I guess I just expected it to work. I'm not expecting perfection, it's free after all, but this is not a small issue.

smartavionics, 2019-05-29:
> but this is not a small issue

To you, maybe.

Jdbye, 2019-05-29:
To anyone with one of these printers. Which I understand isn't the Cura team so I don't expect it to be a priority. I just hoped for at least a little interest in improving their software beyond just what is useful for their own line of printers. But this is not really the place for that, so I'm gonna stop now.

andymozza, 2019-05-29:
@jdbye - on windows I could at least it seems run both versions in parallel. I get a new Ultimaker Master folder structure in program files with the new build and a new program files icon for the 'master' build. I can run both. I have three tools defined, but these tools mixing ratios are defined in printers config and not cura as cura can't control reprap M567 commands yet @smartavionics that's another future conversation. I don't have any mixing ratio issues.....can you give us more detail ?

Jdbye, 2019-05-29:
@andymozza I have all 5 (virtual) extruders set up through Cura using the start gcode, and adding them manually to the json, and on the regular version I can switch between them just fine, except the issue with the hotend turning off (which for some reason doesn't happen on all models)

The issue where it's only printing 75/25 happened once I switched to the fork. I switched back, and without changing a single setting, I am able to print in single color again.

But the really weird thing is... I don't see anything obvious in the gcode that would cause it to switch to the 75/25 extruder. I can confirm on my printer LCD that is indeed set to 75/25, and I can change it back on the LCD, but next time I try to print it changes to 75/25 again. In the gcode there is just one "T0" or "T1" command near the top depending on which solid color I'm printing. Unless there's another command that does the same thing, but I couldn't find anything like that on Google or the Marlin docs.

andymozza, 2019-05-30:
> except the issue with the hotend turning off (which for some reason doesn't happen on all models)

This is because cura tries to turn the heater to standby temp ( defined in Cura ) for a nozzle that will not be used again for a while ( can be managed if you set the standby temps to print temps for all tools), but if Cura decides the tool will not be used again in the print at all, it sends a **Tool temp to 0** command. - This is what screws up prints randomly.

Let us know what you find with the comparing gcode.
I guess that all the tools are defined in cura ( otherwise it wouldn't be able to know about them), But you just change the mixing ration externally right ?

Maybe the commands to change the mixing ratio in print start that you mentioned somehow affect the tool selection in a way we have not considered.

What gcode is used to change the mix rations and what firmware version does the printer use?  ( I have different machine ).

Jdbye, 2019-05-30:
> Maybe the commands to change the mixing ratio in print start that you mentioned somehow affect the tool selection in a way we have not considered.
It should work just the same way as normal, so it's weird that it doesn't in the fork.

I haven't compared the gcode yet, going to do it later today.

The printer is based on Marlin 1.1.8.
I use this in my start gcode to set the virtual extruders:

```
M163 S1 P1.00 ; Left 100%
M163 S0 P0.00 ; Right 0%
M164 S0 ; Save

M163 S1 P0.00 ; Left 0%
M163 S0 P1.00 ; Right 100%
M164 S1 ; Save

M163 S1 P0.25 ; Left 25%
M163 S0 P0.75 ; Right 75%
M164 S2 ; Save

M163 S1 P0.50 ; Left 50%
M163 S0 P0.50 ; Right 50%
M164 S3 ; Save

M163 S1 P0.75 ; Left 75%
M163 S0 P0.25 ; Right 25%
M164 S4 ; Save
```

Jdbye, 2019-05-31:
I don't think this issue has got anything to do with the fork. There's something with the way I'm defining the virtual extruders that's causing issues with switching to any other extruder than the first one, even in the normal Cura. Even though I'm just copying instructions for the Geeetech A10M that I found on YouTube. Removing the gcode for the virtual extruders makes me able to switch between the two normally. Anyone else have Marlin and use virtual extruders? How do you set up the extruders?

andymozza, 2019-06-03:
@smartavionics . Have been using this build with a single extruder machine ( cr10) and also  dual extruder ( single heater). No issues to report for either. I think total 20 small prints on each since getting the build. Thanks, shared heater options seems to be working well for me.

smartavionics, 2019-06-03:
Hi. That's good, Unless you wish for any further changes, I will submit what we have now to the Cura devs for possible inclusion in a future Ultimaker release.

andymozza, 2019-06-03:
Hi, 
I think this is the critical part of it all as it will finally allow these type or printers to complete a print with CURA. I am sure once it gets in to the wild there will be many other suggestions or tweaks etc. Please just submit it.

......Did you raise a bug for that z-hop on nozzle change not showing in the UI? Or do you want me to do that?

smartavionics, 2019-06-03:
> I think this is the critical part of it all as it will finally allow these type or printers to complete a print with CURA. I am sure once it gets in to the wild there will be many other suggestions or tweaks etc. Please just submit it.

Done that, now we have to see if they will accept it or ask for changes.

> Did you raise a bug for that z-hop on nozzle change not showing in the UI? Or do you want me to do that?

I can't actually remember the detail of that but wasn't it a non-issue?

andymozza, 2019-06-07:
Thanks - appreciate the efforts!
The issue with the z-hop on nozzle change was simply that the option doesn't show up until you check another box ( so it was hidden from the UI) but still actually active, even though the nested option made it look like it was all in-active..... so I guess it should be moved out from that menu or something..... It's just a sensibility change really.

LisaErl, 2019-08-04:
May I ask if (and if so, when) this feature will be included in Cura? I know a lot of people with 2in1out hotends who were very happy that Cura 4.2 Beta contained this functionality and are now very sad.

smartavionics, 2019-08-11:
> May I ask if (and if so, when) this feature will be included in Cura? I know a lot of people with 2in1out hotends who were very happy that Cura 4.2 Beta contained this functionality and are now very sad.

This feature has not yet been accepted by Ultimaker so it cannot have been in 4.2 beta. As far as I know, it's only in my builds.

LisaErl, 2019-08-11:
Sorry, then I got something wrong. Someone posted a screenshot of the 4.2 beta with the setting "Shared Heater" in a German forum and so I thought that your modification was included there. And in 4.2 this setting was no longer visible.

smartavionics, 2019-08-11:
Ah, understood. Hopefully it will be accepted into Ultimaker Cura some time soon.

LisaErl, 2019-08-11:
I hope that too

andymozza, 2019-08-12:
@LisaErl - I haven't used main Cura release since @smartavionics first did this work to help us 4 months ago. I see no need to wait.... just use the builds from his dropbox. They are based on the latest 'real' Cura and he has some other good enhancements implemented on top as well.

LisaErl, 2019-08-12:
@andymozza I don't care about me.  I have made postprocessing scripts and have no problems at the moment. But I am in FB groups for A10M/A20M and several times a week desperate owners report and suspect technical errors with their printers, because suddenly no filament comes out of the nozzle. Sometimes they don't even notice that the target temperature of the Hotend is 0. And some find it hard to implement simple Search&Replace Postprocessing Scripts. For these people, I would like to have a solution implemented directly in Cura.

andymozza, 2019-08-12:
I understand and having it main Cura would be great, - but what I have been doing with another brand of printers forums and chat is direct them to the drobox link above, or maybe you could even create a sticky on the forum directing them to use this in the meantime while waiting for main Cura: https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

They still get real latest version cura with all the newest released stuff 'and' smartavionics changes. He keeps it up to date with all the upstream cura changes as well. His release notes tell you what is extra.