smartavionics/Cura #9: Windows builds?

extesy, 2019-06-08:
I would really love to use Cura with all your patches applied, but building everything from source on Windows is very complicated. If it's not too much inconvenience for you, can you please publish Windows builds of your `mb-master` branch from time to time (like monthly or so)? Thank you!

smartavionics, 2019-06-08:
Hello @extesy , the good news is that I already do publish builds for Linux and WIndows. You can find them at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0.

Please note that although I don't intentionally post bad builds they can be buggy because they are based on the upstream master branch + my own developments.

All feedback is welcome, please post to an issue here. Thanks.

smartavionics, 2019-07-28:
Just tidying up.

closed 2019-07-28 by smartavionics