smartavionics/Cura #10: Upper layers are not printed

olegiy, 2019-06-11:
[image]

[WDI3_Hole 15x20mm.zip](https://github.com/smartavionics/Cura/files/3276566/WDI3_Hole.15x20mm.zip)

smartavionics, 2019-06-11:
Yes, and it doesn't print the bottom skins either.

I am aware of this problem and will think about how it can be solved.

eskeyaar, 2019-09-18:
@smartavionics can we just change the top pattern to lines, to generate the skins at top and bottom? 
[image].
this is was done by just changing the top and bottom pattern to lines.

smartavionics, 2019-09-19:
Sure, if it works, do it! There is still a bug somewhere because even with the fill gaps enabled, it will leave that region unfilled when the pattern is zig-zag. I don't think the bug is specific to my releases.