smartavionics/Cura #11: Single heater multi extrusion enhancements

MalikDrako, 2019-07-11:
Thank you for your work on single nozzle multimaterial printing. I've been experimenting with the Palette 2 on a Prusa Mk3S and your 20190618 build and had a couple enhancement requests:

When using a Shared Heater printer, each tool after the starting one needs to be purged. (Maybe only do a brim to the start of the next used material in the tower instead of doing the entire thing with the brim extruder?)

Add an option for minimum extrusion length per tool, and possibly a different one for the starting tool. The Palette 2 requires the first filament section be at least 14cm long, and all subsequent sections be at least 8cm long. If the model plus purge uses too little, the purge should be increased for that tool change. (Easiest might be leaving enough room for the max in the prime tower)

Don't fill in the entire section of the outer layer of the purge tower if that tool is not being swapped to in that layer

I thought I saw that this or an earlier build removed extra temperature changes, but I'm still seeing 10 degree drops when printing with two same-temperature materials. I'm not sure what they correspond to, the printing temperature for both materials is 245 and the standby temperature is 200. I did have one material set with a printing temperature of 235, but it was unused in this print. If I'm remembering correctly, an earlier print that did use that material went down to 225. (Causing a jam since that was a flexible filament and it couldn't push the higher temp material through)

smartavionics, 2019-07-11:
Hi @MalikDrako , thanks for the feedback. I will think about your requests but I have to say that the multiple extruder stuff is not really my area of expertise so I am not familiar with the parts of cura that does all the priming, etc. Also, I normally use single extruder printers so I am not in a position to test any mods myself.

> When using a Shared Heater printer, each tool after the starting one needs to be purged. (Maybe only do a brim to the start of the next used material in the tower instead of doing the entire thing with the brim extruder?)

Sorry, dumb questions, this is on the first layer, right? And this has to be done for all extruders even if they are not used on the first layer?

As for the temperature drop, if you have an example of this, please zip up both the gcode and the project file and attach to this issue and I will investigate.

smartavionics, 2019-07-11:
Assuming that I can make the changes you're requesting, I will need a project file anyway to test against. Even if I can't print the result, I can check the gcode. So please provide a simple example I can work with. Thanks.

MalikDrako, 2019-07-11:
Does this project file/gcode work? I'm just testing with a pair of calibration cubes at 50% height.
[CFFFP_xyz-calibration-cube_twoMat.zip](https://github.com/smartavionics/Cura/files/3384161/CFFFP_xyz-calibration-cube_twoMat.zip)

> Sorry, dumb questions, this is on the first layer, right? And this has to be done for all extruders even if they are not used on the first layer?
Each extruder (except the starting one) needs to be purged the first time it is used regardless of what layer that happens on. If it is the first layer, stop the inner purge tower brim from the previous extruder and start one from that extruder.

I had luck adapting [this tutorial](https://www.thingiverse.com/thing:3278326) for Cura before I got the Palette

smartavionics, 2019-07-11:
The unwanted cooling down is because your initial printing temperatures are lower than the printing temperatures for each nozzle. The initial printing temperature setting is hidden when the heaters are shared but the value is still getting used. That's a bug and I will fix the slicer to ignore the initial printing temps when the heaters are shared. In the meantime, temporarily disable the shared heater setting so that you can edit the initial temps and set them to the required printing temp (i.e. 245). Having done that, re-enabled the shared heater setting.

MalikDrako, 2019-07-12:
One more idea: Target Position, which I think is the percent in the purge tower that it actually changes the tool.
In Chroma, this is the setting description:

> The target of the actual color change within the transition. Setting this around the middle gives the best chance of maintaining calibration, while a slightly earlier value ensures fuller purging. The default (40%) is recommended unless frequent transition bleed occurs.

smartavionics, 2019-07-12:
I still don't really understand the required behaviour.
What happens when an extruder is not used on every layer? does the purge tower still get built?  If only one extruder gets used on a layer does the purge tower still get built? 

Can you provide a project file for a simple example that uses multiple extruders but not on every layer?

Thanks.

MalikDrako, 2019-07-12:
If an extruder (except for the first one that is used to give the tower its structure) is not used on the first layer, keep its portion of the purge tower the same, except add a purge the first time the tool is used.
If an extruder (except the first) is used on the first layer, don't fill in the inner brim with the first extruder and purge the first time that tool is used as above
If an extruder is no longer used, it is currently not included in the purge tower after its last layer, as expected.

This simulates what I'm talking about with a different brim material, red would be extruder 1 and white would be extruder 2
[image]

smartavionics, 2019-07-12:
I'm doing some experiments, just trying to get a feel for what may be useful. Here, for example, you can see that the prime tower lines have been reduced to simulate the effect of limiting the amount of material printed in the tower.  

[image]

I am assuming you still require a min amount of prime tower to be extruded even when the model uses more than 8cm of filament which at the moment is controlled by the "prime tower min volume" setting. That setting (which is specified in mm^3 rather than as a length) effectively sets the number of circles for each extruder in the prime tower so it's really acting more like a maximum value rather than a minimum at the moment but I think we maybe should change its role so that it really does just specify the lower limit on the number of circles that will be printed irrespective of how much filament is used for the lines in the model that get printed next.

We then need a new setting for each extruder: "extruder min volume" which specifies the min volume per layer (which is the combination of volume required for the model and the prime tower circles).  it would typically take a value based on the 8cm length you mention so it would be something like 80 * PI * 0.875^2 == 192.4, say 200 (mm^3). This then would control the max number of circles in the prime tower because if the model uses almost no filament the entire volume of the segment will have to go into the prime tower. The whole tower would obviously have to be big enough to hold the required max volume.

Does this make sense? Thoughts?

MalikDrako, 2019-07-12:
That sounds like exactly what it needs. Just two notes:
That 8cm length is per use of an extruder, not necessarily per layer - dual material prints switch once per layer (probably not a huge deal if it's easier to not account for this and just make it per layer)
The purge volumes I've been testing are getting pretty close to 8cm of filament, I can see it being above the 8cm for some people.

smartavionics, 2019-07-12:
Here's some gcode that has the max number of prime tower circles based on a min per layer volume of 200mm^3 and each layer contains sufficient prime tower circles to make up the required volume.

[CFFFP_prime_tower_test_1.gcode.txt](https://github.com/smartavionics/Cura/files/3388168/CFFFP_prime_tower_test_1.gcode.txt)

[image]

I'm finished for tonight, will probably be working on it again tomorrow evening, cheers!

smartavionics, 2019-07-13:
Hi @MalikDrako , I have something that is probably worth testing. Are you using Linux or Windows?

MalikDrako, 2019-07-13:
I use Windows. I can run manual tests this weekend, but I won't be able to run a proper test until Monday. 
(On another note, I finally got the Palette calibrated last night :) )

smartavionics, 2019-07-13:
OK, I'll make a build so that you can at least see what I have done so far. Will be available in dropbox in a hour or 2.

I have added two settings:

Extruder min volume - this value should be based on the min extrusion length per segment, i.e. 8cm which is approx 200 mm^3.

Prime outermost extruder every layer - defaults to true which is the behaviour up till now.

Here's what the prime tower looks like for your example project when these options are used, notice how near the top of the print the prime tower volume used reduces as the model uses more.

[image]

MalikDrako, 2019-07-14:
Did I grab the right version? I'm not seeing those settings
Dropbox: https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0
Executable: Cura-mb-master-win64-20190713.exe

smartavionics, 2019-07-14:
Hi, even if you select the expert setting visibility in the hamburger menu on the right of the search settings box?

MalikDrako, 2019-07-14:
That was it, I swear I had it set to show all before...

Any idea what would cause Cura to put a single line of the last extruder around the prime tower even though that extruder is never used?
[image]

I also wonder about disabling "Prime outermost extruder every layer", does that mean each extruder is only primed every other layer? If so, I'd be worried about the tower being too weak. What if you changed it to be minimum lines per layer for outermost extruder? That would allow you to keep some structure for the tower without wasting too much filament

smartavionics, 2019-07-14:
> I also wonder about disabling "Prime outermost extruder every layer", does that mean each extruder is only primed every other layer? If so, I'd be worried about the tower being too weak. What if you [...]

That's not a bad idea although I still don't see how the prime tower integrity is ensured when you have prints that don't use all the extruders on every layer. In particular if that extruder that is used to print the outermost part of the prime tower is not used for some layers, it doesn't need to prime at all. Are you saying that it should always prime the outermost extruder even when it isn't used on a particular layer?

MalikDrako, 2019-07-14:
> Are you saying that it should always prime the outermost extruder even when it isn't used on a particular layer?
That seems to be what it's been doing, although it uses the full purge volume when only a couple of lines would be enough. I'll add it to the list of things to test on Monday.

smartavionics, 2019-07-14:
No urgency but could you please send me a project file that has a model that doesn't use all extruders on every layer? Thanks.

smartavionics, 2019-07-14:
I have made a couple of changes and a new build is available:

1 - it now uses the min lines per layer for the outermost extruder when a switch hasn't just occurred.

2 - I have removed some unnecessary travel moves.

MalikDrako, 2019-07-14:
> No urgency but could you please send me a project file that has a model that doesn't use all extruders on every layer? Thanks.

So far, I've just been playing with calibration cubes while trying to get the Palette calibrated. Here's one with different sized cubes that is using extruder 4 when it shouldn't, only 1 and 2 should be used
[CFFFP_xyzCalibration_cube_mm_test.zip](https://github.com/smartavionics/Cura/files/3390115/CFFFP_xyzCalibration_cube_mm_test.zip)

I'll take a look at that build in a bit

MalikDrako, 2019-07-15:
Just tried the build on this computer, and I'm not seeing the new settings. I tried "Expert" and "Show all Settings" and also reinstalling. But the slice is freezing with this in stderr.log: "Trying to retrieve setting with no value given: 'prime_outermost_extruder_always'"

smartavionics, 2019-07-15:
Ok, I'll check it out and get back to you...

MalikDrako, 2019-07-15:
I also tried the 712 build and I'm not seeing the new settings from that, but I did see them on my other computer.

smartavionics, 2019-07-15:
Just checked my windows machine and the cura on that definitely has those settings visible and that version of cura is what's in the latest dropbox file. Could it be the install somehow didn't replace all the previous files?

MalikDrako, 2019-07-15:
Possibly, but I did try deleting the contents of the install folder. (My computer didn't like the uninstaller...)

smartavionics, 2019-07-15:
The other possibility is that dropbox screwed up? I will delete the file on dropbox and add it again.

MalikDrako, 2019-07-15:
I just completed a color bleed test and got somewhere between 250-300 mm^3 for a full purge (red -> transparent)
~140 was enough for red -> gray

MalikDrako, 2019-07-16:
I also tried resetting my settings and the new ones are still not visible.
Do you know where the files for the UI end up?

smartavionics, 2019-07-16:
> I also tried resetting my settings and the new ones are still not visible.
Do you know where the files for the UI end up?

Sorry, I don't for Windows installations and I'm just about to go out for a couple of hours. Can check later and let you know. It's going to be somewhere in the per-user folders.

MalikDrako, 2019-07-16:
I see cura folders in %localappdata% (cache) and %appdata% (profiles, etc), but nothing that looks like UI, any idea what types I should be looking for?

MalikDrako, 2019-07-17:
Here is a project/gcode for 4 materials that shows two issues we haven't gotten to yet. Extruders 2 and 3 (the middle 2) don't reach the 8cm minimum on the first layer. Extruder 1 does because it is used for the brim, and extruder 4 does because it is the first used on the 2nd layer. None of the additional extruders get purged on the first layer.
[CFFFP_xyz-calib-cube_fourMat_1.zip](https://github.com/smartavionics/Cura/files/3399492/CFFFP_xyz-calib-cube_fourMat_1.zip)
And a photoshop showing the proposed solution for the first layer. If an extruder is used on the first layer, stop the inner brim from the previous extruder and continue it with the new extruder
[image]

smartavionics, 2019-07-18:
Hi, I have added a new setting (Prime All Extruders On Initial Layer) that should give you something like you show above. I have also fixed a bug that would put a double brim on the prime tower if you enabled the prime tower brim and the platform adhesion was also set to brim.

A new build is currently being uploaded to dropbox, should be there soon.

MalikDrako, 2019-07-18:
Looks good, I just see one potential issue with it: If an extruder is not used on the first layer (whether or not it is used in the print), there is a gap between the previous and next prime areas. I suggest filling in that area with either the first/brim extruder or extending the previous one to cover that area.
I might also consider changing the name of that to "Prime all extruders on first use"
[image]

smartavionics, 2019-07-18:
>  I suggest filling in that area with either the first/brim extruder or extending the previous one to cover that area.

That's actually harder to do than you may imagine because the prime regions that each extruder use are "set in stone". There's not much flexibility there. I think you need to actually do some printing, especially with tall prime towers to see if there really is a need for that.

I am considering removing that new setting that stops the outermost extruder priming on every layer but will keep it until it has been tried and proven not to be useful.

MalikDrako, 2019-07-18:
What section deals with the prime regions? I'm kind of curious to see what it is actually doing there.

smartavionics, 2019-07-18:
Most of the prime tower code is in PrimeTower.cpp but there's also related code in LayerPlan.cpp and FffGcodeWriter.cpp.

MalikDrako, 2019-07-31:
Now I'm running into a new problem, If the inner sections of the tower skip layers, they tend to stick to the nozzle rather than staying inside the prime tower. One random idea for fixing it, would it be possible to run 2-4 strings through the center of the tower? Either every layer or for 1-2 layers before a floating section

smartavionics, 2019-08-01:
Hi, sorry for the slow response. Do you have a sample project that shows this issue? Thanks.

MalikDrako, 2019-08-01:
The print I ran into the issue on is proprietary, but if I'm understanding the issue right this will do the same thing. I'm testing it now.
[CFFFP_tower_test.zip](https://github.com/smartavionics/Cura/files/3458150/CFFFP_tower_test.zip)

smartavionics, 2019-08-02:
I'm on holiday for a week now so can't look at this until I return.

MalikDrako, 2019-08-02:
No problem, I'll keep investigating in the meantime.

I updated to 20190731 (from 20190718 I believe) and turned off Prime Tower Brim, and the over extrusion issue went away

smartavionics, 2019-08-18:
There's a new build, 0818, that adds a Compact Prime Tower setting. Enabling that removes the gaps from prime tower and simply works from outside inwards on each layer. Let me know if it helps.