smartavionics/Cura #13: V20190724 - Extrusion problems

amagr0, 2019-07-26:
**Application Version** - CURA Master - V20190724
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

Hello @smartavionics,

Lately I'm having some printing problems, some underextrusions issues that is seems to be related with the lack of pressure in the nozzle, due to some irregular continuous path extrusions, like for example this one in the images bellow. 

[image]

[image]

[image]

[image]

[image]

[V20190724_Extrusion_problems.zip](https://github.com/smartavionics/Cura/files/3435384/V20190724_Extrusion_problems.zip)

Thanks

smartavionics, 2019-07-26:
I think what you are seeing in those images is simply caused by the wall width being slightly less in one location and so the inner wall line is splitting into multiple polygons rather than just staying as one polygon. If you reduce the inner wall line width sufficiently it doesn't do that.

A bigger problem is that your profile does not use wall overlap compensation or set a minimum wall flow - both are required to achieve reasonable results when the wall width is an odd number of line widths wide as is the case in your model. If you don't use overlap compensation, it will print two wall lines on top of each other and if you don't set a min wall flow when compensation is enabled, it will print lines with low flows which will lead to underextrusion on the lines that come afterwards.

amagr0, 2019-07-26:
Hello @smartavionics,

But in official Cura version (4.1) this situation doesn't happen, as you can see in the image below.
[image]

About overlap compensation, from my experience it will give me underextrusion problems in most of the situations.

smartavionics, 2019-07-26:
Have you tried 4.2-beta? My build is based on that, not 4.1.

amagr0, 2019-07-26:
No, but if 4.2-beta is giving me this issues I will stay in 4.1.