smartavionics/Cura #14: 20190727 - Ender 3 profile unable to slice with supports - z-hop speed and <blank> error.

JohnEdwa, 2019-07-28:
**Application Version**
Master 20190727 Windows 7 x64

**Steps to Reproduce**
* Add "Creality Ender 3" printer
* Enable supports
* Try to slice - "Unable to slice: The following settings have errors: Z Hop Speed"
* Reduce Z Hop Speed from 10 to 5 and/or disable Z Hop When Retracted
* Try to slice - "Unable to slice: The following settings have errors: "

smartavionics, 2019-07-28:
I did set the default max z-hop speed to a low value (was speed of light!) Please try today's release which sets the max z-hope speed back to speed of light and should allow the slicing to happen. The actual z-hop speed used will be the value you give for Z Hop Speed. Sorry for the inconvenience but if they hadn't screwed around with it in the first place all this z-hop nonsense would never have occured!

JohnEdwa, 2019-07-28:
0727 is the latest Windows build, unless it's still uploading and isn't showing yet?

smartavionics, 2019-07-28:
Oh, you're probably right. I will check and upload if required. Broadband's a bit slow so it will take around 30 mins, sorry.

smartavionics, 2019-07-28:
Uploaded now, give it a go!

JohnEdwa, 2019-07-28:
Nope, still there. 
Tested it with some other printers as well, Ultimaker 2+ works fine as does a Tevo Tarantula, but Creality Ender 3, 5 and CR-10 all have the issue still. They default to Z hop 10 which shows as invalid and 5 is still the highest it can go, and fixing that still leaves the one blank error.

smartavionics, 2019-07-28:
I think that's because they have incorporated the crawesome mod and all the Creality profiles and machine definitions have changed. People have been griping about it after trying the beta.

The crawesome base printer definition contains this:

```
        "machine_max_feedrate_z": { "value": 5 },
```

IMHO, the best solution is to install the printer settings plugin from the marketplace and then simply edit the max z speed setting and set it to something larger.

smartavionics, 2019-07-28:
You would of course have to edit it for every profile (I think).

JohnEdwa, 2019-07-28:
It still leaves the blank empty error that isn't connected to any value - nothing shows up red when I enable "show all settings", I just get a popup at the bottom and it refuses to slice.

[image]

smartavionics, 2019-07-28:
Yes, I got that until I increased the value of machine_max_feedrate_z to be > than the z-hop speed.

smartavionics, 2019-07-28:
Weirdly, with z hop speed set to 5. I can increase the max z speed from 5 and it will keep giving that error you show above until the max speed reaches 10 and then it lets it slice. Don't understand that at all.

JohnEdwa, 2019-07-28:
Not larger than Z Hop Speed, it needs to be 10 or higher regardless of the hop. Z Hop 1mm/s -> fails to slice. Maximum Z Speed 9mm/s -> fails to slice. Max Z Speed 10 -> Success.

Why, IDK. But that fixes it.

smartavionics, 2019-07-28:
It's an upstream bug, if you want to open an issue, please do that at Ultimaker/Cura. I'll close this issue but you can still comment on it if you wish.  Cheers!

closed 2019-07-28 by smartavionics

smartavionics, 2019-07-28:
See https://github.com/Ultimaker/Cura/issues/6073.