smartavionics/Cura #15: [FR] Better brims for supports.

JohnEdwa, 2019-08-02:
[image]

The short version is that because Cura likes to generate thin but tall supports, they rarely stay attatched without also enabling a brim for the whole print. For some reason, the people at Ultimaker haven't really been keen on adding anything to help this (it took them three years to add a brim to the prime tower and that is often *wider* than these supports), so I turn to you.

I would love to see at least one of these, or something else if you can think of a better solution:

* Adding a "Support Brim Only" bed adhesion that doesn't generate a brim for the model, only for the supports.

* Make support brims ignore "Brim Distance", so I can generate a brim that doesn't actually touch the model, but that anchors the supports. Possibly tied to the "Brim Replaces Supports" setting, if it makes sense to do so.

* Enforcing the "Support Brim Width" so that if the support (and therefore the inside brim) is narrower than the brim width, the brim also expands outside until it is big enough. This would make sure the contact area is always at least that size even for tiny supports even if I use a skirt.

* And a wildcard, making regular supports work like the Tree supports so that they are wider at the bottom with a certain configurable width to height ratio, so that Cura would stop generating these super thin tall supports with no adhesion.

AbeFM, 2019-08-03:
[image]
Yeah, the option is there, just ignored. No brims on support UNLESS it's on for the whole model, and turning off the check box does not remove supports if they are on for the whole model.
Brim line count also seems to be ignored.

JohnEdwa, 2019-08-03:
The support brim is created inside/under the support if enabled, replacing the first layer which is normally a a grid. So it's kinda working as intended, but it doesn't help when the support is generated uper thin but tall. 

>You can try "conical supports", under experimental, they are wider at the bottom. I was going to suggest you use tree supports for the example shown

Oh neat, that's exactly what I want. How long has that been hiding in there?
Though it could still save some filament if it used an outside brim, bu it works perfectly.