smartavionics/Cura #17: [FR] Adding an "Infill Fan Speed" option (for stronger layer adhesion)

JohnEdwa, 2019-08-14:
Stefan from CNC kitchen recently did a video on how the part cooling can have a tremendous impact on the layer adhesion with PLA. The full video is 16 minutes, [but the relevant graph is here](https://youtu.be/Bl2ESvtBiLo?t=491); basically, the difference between 100% fan and 0% fan was a pretty linear increase of up to 50% - the less cooling, the stronger the layer adhesion. The downside for normal printing is obviously worse overhangs and print quality in general.

If there was a way to reduce or stop the part cooling fan only when infill is being printed, I would assume that would simply increase the strength of the part with practically no degradation of the overall visual quality. 

And combining it with the Overhanging Wall Fan Speed, Supported Skin Fan Speed and Minimum Layer Time, it might be possible to drastically increase the layer adhesion by only using the cooling when it's necessary for the visual quality, and otherwise print with basically no cooling at all, giving the best layer adhesion.

smartavionics, 2019-08-14:
Hi. That's a very interesting idea. It's not hard to do so I will put something together. My Cura has some mods that make the fan speed changes more accurate compared to the stock Cura as it takes into account the lag for the fan to spin up/down. I will let you know when I have something to try. Thanks.

smartavionics, 2019-08-15:
Hi. I have something you can try. There are new infill settings: Infill Fan Speed Override and Infill Fan Speed.
I am just uploading a Windows build (takes around 30 mins) and a Linux build will follow.

JohnEdwa, 2019-08-18:
It doesn't seem to work in the 20190815 windows build at least, searching for "Infill Fan" or other similar things doesn't show up anything.

smartavionics, 2019-08-18:
Strange. I'm just uploading another build anyway and I checked that and it definitely has the new infill fan settings. Try 0818 when it arrives.

closed 2019-12-23 by JohnEdwa