smartavionics/Cura #18: Filament Start Code Dialog Errors

AbeFM, 2019-08-15:
**Application Version**
802 win10

**Printer**
Custom printer for HicTOP D3 Hero (idex)

**Steps to Reproduce**
Edited extruder specific start g-code, exited, used.
Returned to printer settings, selected text, hit backspace, for both extruders, exited.

Next prints had start up code inserted only for E0, E1 was not using it.

Loaded CURA, checked box only to find it blank. In frustration, I clicked in the text box once and the previous g-code reappeared.

I selected it again, and hit delete (backspace?), and said ok.

Since then it's been ok.

**Expected results**
Text to be gone when it is not displayed

**Additional Information**
Kinda just putting it out there in case someone else sees it. Unsure if it's somehow related to your branch relative to the main one.

Probably best ignored, I'll post if it comes up again.

smartavionics, 2019-08-15:
I haven't made any changes to this part of Cura but there are differences in how my releases are build (e.g. different version of Qt used). If it happens again, could you see if the same problem occurs when using Ultimaker's build? We'll leave this issue open for now.

AbeFM, 2019-08-15:
I will try to reproduce this (and the minimum area bug from other thread)
in the release versions and report back. Visiting family in town for a
week, but I'll try. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2019-08-15:
[image]

First Click:
[image]

Printer is doing it again, E0 uses start code, E1 does not.

I tried overwriting with " ;Trying to overwrite undesired startup code E1" and a line feed. We'll see what it does, but I'm late for a wedding so I'll have to come back to it. :-) Still haven't compared to official version, sorry. Without supports working there's little reason to use it.