smartavionics/Cura #19: Getting Brims when unexpected, double extrusion on prime tower brim

AbeFM, 2019-08-22:
**Application Version**
0802  - may happen on official CURA, haven't checked yet

**Platform**
win10-64

**Printer**
Generic FFF IDEX 

**Steps to Reproduce**
Turn off Support Brim, Prime Tower Brim, etc, but keep Brim for model adhesion. Used round tower, don't know if it matters.

**Actual Results**
Tower and support get brim (I think model, also)
[image]

**Expected results**
No brims should be present.

**Additional Information**
If you have Prime Tower Brim on at the same time as model Brim, the extrusion used for the Prime Tower brim is ~2x (I tested by tweaking flow settings). It's massively over-extruded and impacts Prime Tower adhesion.
The printing looks fine in preview, but when printed you get results as shown:
[image]
The front two are printed with "Prime Tower Brim" turned off.

I would expect, if there were an issue, that the brim would be printed twice, but this is not the case. It is one pass with over-extrusion, none of the other brims/bottoms show this effect - though I think it might happen on the interior of the prime tower as well.

smartavionics, 2019-08-22:
I have done some work on the prime tower recently but can't find anything to explain what you are seeing. Please provide the gcode that produces the bad results. Thanks.

smartavionics, 2019-08-23:
OK, I have found an issue in the upstream code and have submitted a PR to them for that. I will make a new build later today with that fix.

AbeFM, 2019-08-23:
Groovy. :-)
Sorry for posting stuff that's probably not your fault. It takes ~12 months to get an answer on CURA forums, so I almost have to check here first to get an answer. Glad it worked out.

Thanks!

smartavionics, 2019-08-23:
A new build is available. It contains a fix for this issue so I closing it.

closed 2019-08-23 by smartavionics