smartavionics/Cura #20: Edit readme - Include Dropbox link

AbeFM, 2019-08-23:
Not sure if you want to do this, but:
There's pre-built releases for Linux and Windows available from https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

That would save some time for folks doing testing. :-)

smartavionics, 2019-08-23:
I don't understand how this helps because to get to the readme people have to access the folder anyway. Don't they?

AbeFM, 2019-08-23:
[image]
Clicking the "CODE" button, the first screen that comes up on the project, displays the readme. That front page is where I assume (right or wrong) the most important information will be, including how to access.

smartavionics, 2019-08-24:
Oh, that readme file in the repo. Now I understand.