smartavionics/Cura #22: [FR] Add "standby temperature" to the visible settings list

extesy, 2019-09-06:
**Application Version**
Cura 20190904 from your dropbox

**Platform**
Windows 10 x64

**Printer**
Wanhao Duplicator i3 Plus

**Steps to Reproduce**
1. Select "show all settings" in the settings panel
2. Try to find "standby temperature" (see screenshot)

**Actual Results**
"Standby temperature" is not visible. By some reason it only exists in the settings menu > material profiles, but not on the settings panel.

**Expected results**
Be able to override "standby temperature" from the panel like any other material settings.

**Additional Information**
[image]

smartavionics, 2019-09-06:
Hi, I think the standby temp setting is hidden when you only have 1 extruder enabled.

extesy, 2019-09-06:
Yes, and I don't think that's a right decision because it's used at the beginning of the print as a starting extruder's temperature while it waits for the bed to heat up. Even for a single extruder.

Jdbye, 2019-10-19:
> 
> 
> Yes, and I don't think that's a right decision because it's used at the beginning of the print as a starting extruder's temperature while it waits for the bed to heat up. Even for a single extruder.

Easily solvable by adding heating gcode to your start gcode. I've had it set up that way for a long time, because I wanted the bed and extruder to heat up at the same time, while the automatic bed leveling is running. Saves at least half the time it takes to do the pre-print routines.