smartavionics/Cura #23: [FR] Add a toggle to disable combing in initial layer

extesy, 2019-09-07:
**Application Version**
Cura 20190904 from your dropbox

**Platform**
Windows 10 x64

**Printer**
Wanhao Duplicator i3 Plus

**Steps to Reproduce**
Set combing mode to All

**Actual Results**
Combing is enabled in initial layer which sometimes gives pretty bad results on my printer/bed which struggles with adhesion. When I disable combing and the printer switches to z-hop then everything is printing perfectly. This is only a problem for the initial layer.

**Expected results**
Be able to use full combing mode everywhere except for initial layer.

smartavionics, 2019-09-07:
Hello @extesy , could you please provide a sample project file that you have problems with on the first layer. Thanks.

extesy, 2019-09-07:
Sure, here's the today's example: [Hexapod - Base.3mf.zip](https://github.com/smartavionics/Cura/files/3586896/Hexapod.-.Base.3mf.zip) (combing is off)

This problem happens on my printer when the first layer has many small details, like in the attached project. In this case the only thing that helps for me is turning combing off. But I only need this for the first layer and would prefer to have it on for the rest of the print.

I don't think there's anything wrong with Cura or your patches. But having this option would certainly help with my prints.

smartavionics, 2019-09-07:
Thanks for the file. Have you tried this combination? Combing set to "no skin" and max comb distance with no retract set to something like 10. If you haven't tried it or similar yet, please see if that works better.

extesy, 2019-09-07:
Thanks for suggestion! I will try that and report back.

extesy, 2019-10-27:
I had a chance to experiment some more and here's a model where I couldn't get the good first layer until I enabled "no combing" + "z hop when retracted":

[image]

The problem was those 2 groups of 4 small slanted lines. With combing some of these lines didn't stick and/or were pulled by a nozzle. Without combing and with z-hop I got them to stick. Again, this is only a first layer problem.

extesy, 2019-11-02:
Another example is this model - https://www.thingiverse.com/thing:3495390. I can't print the first layer without disabling combing.

smartavionics, 2019-11-03:
Hello @extesy , I have added settings to disable combing and force z-hops only on the initial layer. I can build a release but note that due to extensive upstream changes, Cura will not load any 3rd party plugins from the marketplace until the plugin authors update their plugins to the latest API. Plugins that UM supply are still available. So if you use 3rd party plugins, my next release is going to be a problem. Do you still want me to make it? It's no hassle if you do.

extesy, 2019-11-03:
Thank you! I can live without plugins until they are updated by the authors, so if you could please compile a new build that would be very useful.

smartavionics, 2019-11-03:
Hi, I made the build anyway, it's already on Dropbox. Cheers!

extesy, 2019-11-20:
Works perfectly, thank you!

closed 2019-11-20 by extesy

smartavionics, 2020-05-05:
Hi @extesy , someone requested the ability to enable z-hops for the first 2 layers so from today (20200505) my releases will now have a setting called Z Hop Initial Layers instead of Z Hop Only On Initial Layer which you may have been using.

extesy, 2020-05-05:
Thanks for a heads-up! I'll update my config accordingly.

5axes, 2020-05-06:
Would be also Great a Z Hop on the Ironing Layer to avoid mark on this particular operation