smartavionics/Cura #24: [FR] Supported skin flow & Supported skin speed

kulfuerst, 2019-10-01:
Hey smartavionics,

it would be brilliant if it was possible to not only select the supported skin fan speed, but also flow and feedrate. Enabling supported skin bridge settings so to speak. I believe this may enable much lower support Z offset and therefore better dimensional accuracy of prints in the Z direction! For prints with larger nozzles it would also make the removal of supports easier which normally gets really difficult. It may also possibly (crossing fingers!) even enable a support Z offset of 0 when printing with a low flowrate while using large nozzles.
Long story short, this could potentially help improve supports a great amount. Thanks a ton for your awesome build :)

Best wishes

smartavionics, 2019-10-02:
I have implemented this and it will appear in the next build. What platform are you using?

kulfuerst, 2019-10-02:
Lightning fast as always. Im running your most recent build on Windows 10. Currently running all kinds of support tests so I really can't wait to check out what additional options this brings to the table :)
Thanks a lot

smartavionics, 2019-10-02:
> Lightning fast as always.

Not really always true but often.

I will do a Windows build so you can try the new settings out, will be available later today.

smartavionics, 2019-10-02:
New Windows build (1002) now on Dropbox.

kulfuerst, 2019-10-02:
Okay, I've run some tests and they are extremely encouraging but also yield one problem that may be difficult to solve.
The images you can see below are the underside of the test print printed with a standard 0.4 mm Nozzle and a Support Z Distance of 0 mm (!!!) after I have removed the supports. The supports were very easy to remove and the underside of the print is as smooth as can be. As you can also see in the screenshot, the Supported Skin Settings only apply to supported skin, not supported walls, and this is causing the issue that supports that are directly under a skin region are easily removable even at a Support Z distanze of 0 mm, but the corresponding wall lines cannot be removed.
I wouldn't be surprised if this is very difficult to fix in the slicer and can understand if this can't be fixed without an unreasonable amount of work. If however, you can think of a fix to this issue that can apply the Supported Skin Settings also to supported walls, I believe this could result in a method of printing supports that are easily removable at a Z distance of 0 mm, yielding better looking undersides as well as more dimensionally accurate prints. My test results are promising.
Thank you for your help and I hope you find this equally fascinating as me.

[image]

[image]

[image]

smartavionics, 2019-10-02:
If you want walls to be treated specially as well you may as well just use the bridge settings. Currently, the presence of support stops the bridge settings being used but I could add a setting that makes it ignore the support so the skins and the walls above the support would be treated as bridges. In that case, the settings I added today are no longer useful and I will remove them again.

kulfuerst, 2019-10-02:
I would be very happy to test that out :)

smartavionics, 2019-10-02:
OK, I've removed the settings I added earlier and introduced a new Bridge Over Support setting that ignores the support as far as bridging is concerned. I just updated the same 1002 file so download it again.

kulfuerst, 2019-10-03:
I think I have spotted a bug in the bridge skin generation and bridge wall generation. As you can see in the images, Cura generates bridge skin in an area where it is actually supported and shouldn't be generated (image 1). It also, the next two layers up, generates bridge walls where they are actually supported (image 2 & 3).
If I select Bridge Has Multiple Layers, Cura tries to cram all the bridge layers into one (image 4).
I'm attaching the STL, gcode and project file below.

[image]

[image]

[image]

[image]

[bridge generation bug.zip](https://github.com/smartavionics/Cura/files/3686063/bridge.generation.bug.zip)

smartavionics, 2019-10-03:
I agree that something weird is happening when the bridge has multiple layers as it appears to overlap skin regions. I will look into that. However, I think the behaviour for a single bridge layer is OK. There really isn't any support under the region where it bridges so the bridge is entirely reasonable, isn't it?

kulfuerst, 2019-10-03:
I agree that from a practical standpoint it may seems like useful behavior in some cases. In my arguably very special use case right now this is not welcome however as I need the top layers to bond very well to the bridge walls. I believe that is behavior that is also generally desirable so the bridge walls become an integral part of the whole print with uniform strength.
Besides the practicality of it, it appears to be in conflict with how bridge walls are defined, as unsupported, and it appears arbitrary too. In plenty of other regions of the printed object, walls placed on top of bridge walls are printed with normal wall settings. The fact that the behavior is unpredictable is making it difficult to use.

kulfuerst, 2019-10-03:
Ohhh there may be a misunderstanding here. I'm talking about Cura putting bridge wall on top of bridge wall. The lower bridge walls are of course absolutely justified. I've got a better image of it here:

[image]

smartavionics, 2019-10-03:
It doesn't for me, as you can see in this image, the wall that is immediately above the outer bridge wall on the layer below is not itself printed as a bridge wall, it's normal wall. The walls that are outside of that wall are bridged but that's because they are above air.

[image]

smartavionics, 2019-10-03:
Ah, perhaps you are talking about the skin lines on the layer above as shown here. I will look into that...

[image]

kulfuerst, 2019-10-03:
That is exactly what I meant! I really should've gone into Line Type view. I thought those were bridge walls, not bridge skin. Unwanted bridge skin also shows up in earlier layers like 51 - 54:

[image]

smartavionics, 2019-10-03:
I think I have fixed this, please try 1003 release and let me know if there are further issues. Thanks.

kulfuerst, 2019-10-03:
That did fix both issues indeed! I'm very impressed.
Btw, in case it is of interest. When Bridge Over Supports in enabled it is possible to print with a Support Z Distance of 0 when using layer heights of 0.2 mm or greater. Important is very low flow (50% or less), density of about 75% and 100% fan speed. Then the supports are easy to remove and the surface finish is brilliant :)
Still working on getting it to work with lower layer height.

closed 2019-10-03 by kulfuerst

smartavionics, 2019-10-03:
Thanks for the info, sounds very promising.