smartavionics/Cura #25: Something changed between 20190910 and 20190914 that made Ender 3 slicing take forever or just fail.

JohnEdwa, 2019-10-01:
**Application Version**
master 20190914
master 20190928

**Platform**
Win 7 x64

**Printer**
Creality Ender 3.
[CE3_square_tower.zip](https://github.com/smartavionics/Cura/files/3677291/CE3_square_tower.zip)

**Steps to Reproduce**
Clean install of 20190914 or 20190928.
Create a Creality Ender 3 printer.
Slice seemingly any model.

**Actual Results**
"Prepare" takes a few seconds, then the "Slicing" bar moves up to around one third and gets stuck.

**Expected results**
Slicing should finish.

**Additional Information**
20190910 works without any issues.
Creality CR-10 works.
Creality Ender 5 works.
Creality CR-20 Pro works.
Ultimaker 2 works.

smartavionics, 2019-10-01:
I tried your project on my current build and it slices with no problems. Nothing of any interest has changed since the 0928 build. Could you please zip up and attach cura.log and stderr.log and stdout.log so I can see if there's any hints in there as to what the problem could be. Thanks.

JohnEdwa, 2019-10-01:
[cura_logs.zip](https://github.com/smartavionics/Cura/files/3677696/cura_logs.zip)

This is with 20190928 after importing the project linked above, trying to slice it twice, then closing the program.

smartavionics, 2019-10-01:
The logs are saying the problem is that the back end can't find a value for the **mesh_temperature_delta setting** which has just been introduced in the 0928 build. It should be defined in the file resources/definitions/fdmprinter.def.json in the Cura installation. Either I goofed and it isn't in that file or, somehow, you are using an earlier version of that file. I just checked and I am sure that setting is in the file so maybe (for whatever reason) you are still using a previous copy of fdmprinter.def.json?

JohnEdwa, 2019-10-01:
Not really sure how that would be possible, as you notice I always install the master to a new folder, and for these tests I removed the old "master" profile folder and renamed all the "4.x" ones so it wouldn't try to import them.

JohnEdwa, 2019-10-01:
Okay, I figured it out, it's because Cura makes cached versions of them in ``%LOCALAPPDATA%\cura`` instead of using the ``%APPDATA%\cura`` which I missed. It actually *did* update fdmprinter, but that's not the broken one, it was the ``creality_ender3`` file.

smartavionics, 2019-10-01:
OK, but do you still have a problem with the creality_ender3 ? That's been changed upstream, not by me.

JohnEdwa, 2019-10-01:
No, it seems to have been a weird upgrade problem that for some reason didn't update the cached version of that file. It was fixed when I removed the cache and let Cura create them from scratch. 

Just to test I wiped everything, created them with 0910 and then upgraded to 0928 and it still works, so whatever it was probably resulted from some very specific order I started the 0910, 0913, 0914 and 0928 builds by accident (at one point I accidentally had one version pinned to my taskbar, one assigned to launch .stl files and a third one being launcehd from F360, so no wonder it got confused :) )

closed 2019-10-01 by JohnEdwa