smartavionics/Cura #26: [FR] Make "Infill Layer Thickness" behave like "Support Z Distance"

kulfuerst, 2019-10-01:
Hello smartavionics,

it would be wonderful if "Infill Layer Thickness" could also round up to the next multiple of the layer thickness, the same way the "Support Z Distance" does. That way it wouldn't need to be changed with every print and one could set it once and never worry about it again. The same way that many people have set the "Support Z Distance" to a value like "0.165" to always have the Z distance be a multiple of the layer thickness until 0.17mm layer height, at which the "Support Z Distance" automatically starts being equal to the layer height.

Currently, the "Infill Layer Thickness" is forbidden to be smaller than the Layer height and only rounds down.

Thank you very much for your consideration 🙂

smartavionics, 2019-10-02:
I think you need to create an issue for this at https://github.com/Ultimaker/Cura/issues. If you do that and they don't reject the idea, I will do what is required to implement it.

kulfuerst, 2019-10-02:
I've already posted this on the [Ultimaker Cura + Plugins forum](https://community.ultimaker.com/topic/28974-fr-make-infill-layer-thickness-behave-like-support-z-distance/?tab=comments#comment-243863) and considered this cross posting bad style already. Got no reply there. Should I still post on the Cura github as well?

smartavionics, 2019-10-02:
Yes, that would be best because then the developers have to reply. The forum they can ignore.

kulfuerst, 2019-10-02:
Feature Request opened here: https://github.com/Ultimaker/Cura/issues/6465
Let's see what they say :)

kulfuerst, 2019-10-02:
A very quick and easy fix was posted there! I've tested it and it works :)
https://github.com/Ultimaker/Cura/commit/672fc58930052749c497862242f7871c888ac223

closed 2019-10-03 by kulfuerst