smartavionics/Cura #27: Request: Bundle common plugins by default

probonopd, 2019-10-10:
Would you be willing to bundle some common plugins, like the OctoPrint plugin, in your binary builds? The Cura team said earlier they won't do this but I'd find it much more convenient to have common plugins to come pre-bundled.

smartavionics, 2019-10-11:
Hi. Possibly, but I can't see any real benefit in doing that. I certainly won't be working on that myself. If you wish to provide a PR, I'll at least consider it.

probonopd, 2019-10-11:
The advantage would be that a user can download the AppImage and have the most popular plugins work out-of-the-box, without having to install them and without having to restart Cura. In case you are interested in having that, I can work on a PR - but only if you like the idea.

smartavionics, 2019-10-11:
I guess, I'm fairly neutral on the idea. Being more positive, if you produce a PR, I'll incorporate it.