smartavionics/Cura #28: Crash in 20191027 build

extesy, 2019-10-27:
Latest 20191027 build crashes on startup, the previous 20191022 build had no problems.

**Application Version**
20191027

**Platform**
Windows 10 x64

**Steps to Reproduce**
Start Cura

**Actual Results**
Crash

**Expected results**
No Crash

**Additional Information**
Traceback:
```
Traceback (most recent call last):
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 998, in event
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 429, in event
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Event.py", line 189, in call
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 162, in setInitialActiveMachine
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 309, in setActiveMachine
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\Settings\ExtruderManager.py", line 383, in fixSingleExtrusionMachineExtruderDefinition
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 106, in createExtruderStackWithDefaultSetup
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\Machines\ContainerNode.py", line 66, in container
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 128, in findInstanceContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Settings\ContainerRegistry.py", line 193, in findContainers
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 212, in emit
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Signal.py", line 315, in __performEmit
  File "C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin\ZOffsetPlugin\ZOffsetPlugin.py", line 43, in _onContainerLoadComplete
    container = ContainerRegistry.getInstance().findContainers(id = container_id)[0]
[... 949 lines cut ...]
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000ec40 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x0000f8f8 (most recent call first):
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 306 in _logInfoWidget
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 156 in _createDialog
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 77 in __init__
  File "C:\Users\burto\cura\cura-build\build\inst\bin\cura_app.py", line 120 in exceptHook
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\UI\CuraSplashScreen.py", line 103 in showMessage
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 503 in showSplashMessage
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 823 in runWithGUI
  File "C:\Users\burto\cura\cura-build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 785 in run
  File "C:\Users\burto\cura\cura-build\build\inst\bin\cura_app.py", line 186 in <module>
  File "Console.py", line 26 in run
  File "__startup__.py", line 14 in run
```

smartavionics, 2019-10-27:
Sorry about that. There's been some major upstream changes. It looks like a problem with the ZOffsetPlugin.

If you remove the folder C:\Users\olega\AppData\Roaming\cura\master\plugins\ZOffsetPlugin does it still crash?

extesy, 2019-10-27:
Yes, that helped. It doesn't crash anymore. I didn't actually use that plugin so I'll just remove it. Thanks for a quick response!

closed 2019-10-27 by extesy