smartavionics/Cura #29: "Small Feature Speed" does not obey "Minimum Speed"

kulfuerst, 2019-11-11:
**Application Version**
20191003

**Platform**
Win10

**Printer**
Anycubic i3 Mega

**Steps to Reproduce**
Set "Small Feature Speed" to 50%, set "Minimum Speed" to certain value (e.g. 15 mm/s)

**Actual Results**
On layers that do not take long to print and that contain small features, the minimum speed will be halved leading to extremely slow feedrates and quality issues.

**Expected results**
The speed does not drop below the value chosen for "Minimum Speed"

**Additional Information**
[image]

[Little MasterSpool Roller Remix.zip](https://github.com/smartavionics/Cura/files/3833184/Little.MasterSpool.Roller.Remix.zip)

smartavionics, 2019-11-11:
Yes, you're quite right. I discovered this myself the other day when I was printing a very small part and wanted slow layers so it didn't overheat. I found it was actually printing at 2.5mm/S. Part came out really nice! Anyway, I'll take a look at not reducing the speed below the min speed.

smartavionics, 2019-11-12:
I've fixed that for my next release and also submitted a PR to UM. Thanks for the feedback.

closed 2019-11-12 by smartavionics

kulfuerst, 2019-11-12:
I believe that, if anyone, you're the one who deserves all the "thanks" :)