smartavionics/Cura #30: [FR] Enable per-mesh bed adhesion settings

extesy, 2019-11-20:
Would it be possible to configure brim/raft settings per mesh when there are multiple objects in the scene? For example, if I print 5 objects at a time but only one of them needs brim - currently there is not way to do that. I have to enable brims for entire print. But this completely messes up objects where first layer is a complex concave/zigzagging are because brim completely closes tiny gaps between lines and it's almost impossible to remove it.

I don't know if this is a fundamental issue of the slicer design but there are a lot of per-mesh settings (support, material, infill, etc) _except_ bed adhesion.

smartavionics, 2019-11-24:
Hi, sorry for the late response. My Cura build for today (1124) allows you to set the Brim Distance value per-mesh. That was easy to implement and I hope it is sufficient for your needs. I think it would be quite a lot more work to be able to have some meshes with skirt and others with brim.

[image]

extesy, 2019-11-24:
I guess this works as a workaround if I specify the brim distance to be large enough to disconnect it from the model. Thank you!

I'm just curious - why is it so hard to disable brim per mesh? Is it because of some assumptions built in CuraEngine?

smartavionics, 2019-11-24:
> I guess this works as a workaround if I specify the brim distance to be large enough to disconnect it from the model. Thank you!

The whole point of the brim distance is to provide a small amount of separation between the brim and the model. Normally, I use less than a line width but there's no reason why you can't use a much larger distance to keep the brim well away from the model.

> I'm just curious - why is it so hard to disable brim per mesh? Is it because of some assumptions built in CuraEngine?

It would be possible to disable brim per mesh quite easily with a per-mesh setting so that a mesh either has a brim or nothing. Putting a skirt around a mesh that doesn't have a brim could be done but at the moment the engine doesn't use individual meshes when creating the skirt shape it uses the hull formed from all of the meshes. I can imagine it getting complex if some meshes have brim and the others have skirt. Would it still try and put the skirt around groups of meshes?

extesy, 2019-11-24:
Interesting, I didn't think of the case when skirts and brims are mixed. What I had in mind is a simple on/off switch for whatever bed adhesion method is globally configured. With the current workaround (brim distance) printing completely disconnected brim is just a waste of filament, although it's still useful in some circumstances. Being able to completely disable brim per mesh would be ideal.

smartavionics, 2019-11-24:
> Being able to completely disable brim per mesh would be ideal.

Sure, but that's yet another setting. One of the reason that the Cura UI is so sluggish is because there are far too many settings. These days, I am reluctant to add new settings unless there is no other sensible way of achieving the same goal.

extesy, 2019-11-24:
Technically that's not a new setting, that is an existing setting "Build Plate Adhesion Type" with the value set to "None".

smartavionics, 2019-11-24:
Good point but as you can't give an individual mesh either the raft or skirt settings without doing a lot of work in the slicer it's not possible to make the adhesion type mesh specific.

smartavionics, 2019-11-26:
Hi, I found a problem with the brim distance feature on the 1124 release, please use 1125 instead. Sorry for the inconvenience.