smartavionics/Cura #31: Python error

eskeyaar, 2019-11-25:
**Application Version**

Cura-mb-master-win64-20191124

**Platform**

Windows 64bit
**Printer**

**Steps to Reproduce**

Just downloaded and installed and overwritten on previous version win64 20190912
**Actual Results**

[image]

**Expected results**

Should open normal as before
**Additional Information**

Previous version: Cura-mb-master-win64-20190912

smartavionics, 2019-11-25:
From the README file that you should read...
> * 20191012 - Windows build using PyQt 5.13. **CAUTION: don't install on top of previous mb-master release, uninstall the old release first!**

eskeyaar, 2019-11-25:
oops. Missed it. 
Thanks again.

closed 2019-11-25 by eskeyaar