smartavionics/Cura #32: python error on start

baikal, 2019-12-27:
**Application Version**
Cura-mb-master-win64-20191226

**Platform**
Window 10 64bit, build 18363.535 (1909)

**Steps to Reproduce**
* uninstalled old Cura-mb-master, removed files from %AppData%/cura (as of readme)
* installed Cura-mb-master-win64-20191226.exe

==> python error on start

**Additional Information**

[image]

smartavionics, 2019-12-27:
Hello @baikal , sorry about that. I don't know what's causing the problem as that release works OK on my build machine. I can see from the error messages that it is related to new functionality that has recently been added upstream by the UM developers. Can you please locate the stderr.log file and attach that as it may contain some useful info. Thanks.

baikal, 2019-12-27:
Hi @smartavionics 
I'm sorry, I could not find stderr.log (nor a 'master' directory under %AppData%/cura as it was created by former versions of mb-master) ...

smartavionics, 2019-12-27:
The stderr.log file should be in the directory above where that master directory is created.

I can't reproduce the crash here but I do have a thought as to what is going wrong so I shall make a new build tomorrow that may work better. I'll reply here again when it's ready.

smartavionics, 2019-12-28:
I am just putting a 1228 release on dropbox, please try it. I still can't reproduce the crash here.

baikal, 2019-12-28:
this is strange - yesterday after uninstalling 20191226, installing 20190704 (had that installer lying around), running 20190704 once, uninstalling 20190704 (this time without deleting %appdata%/cura/master) and then installing 20191226 again - no more error.

downloading 1228 as of writing this  ...

baikal, 2019-12-28:
installed 20191228 'over' 20191226, everything is working fine.

thanks for all your efforts, sorry for the fuzz.

smartavionics, 2019-12-28:
No problem, let's hope you don't see that occurring again.

closed 2019-12-28 by smartavionics