smartavionics/Cura #33: Settings scrambled

kulfuerst, 2019-12-31:
**Application Version**
20191228

**Platform**
Windows 10

**Printer**
Anycubic i3 Mega, project file and screenshots attached

**Steps to Reproduce**
I don't know. I have disabled all plugins however, but that didn't change anything, leading me to believe that plugins are not at fault here.

**Actual Results**
Settings get scrambled. making changes to the settings no longer possible.

**Expected results**
Behavior as normal

**Additional Information**
Error already occurred with 20191128

[scrambled text.zip](https://github.com/smartavionics/Cura/files/4012502/scrambled.text.zip)

smartavionics, 2019-12-31:
Yeah, I have seen that myself. Does collapsing the setting group and opening again fix it?

kulfuerst, 2019-12-31:
I found that the issue can temporarily be fixed by expanding the section above the one that is scrambled. It's like it resets the layout and cura then places everything correctly.

closed 2020-03-21 by kulfuerst