smartavionics/Cura #34: [BUG] Cannot change values of certain settings

kulfuerst, 2020-01-02:
**Application Version**
20191228

**Platform**
Windows 10

**Printer**
Anycubic i3 Mega

**Steps to Reproduce**
Load profile and try to change certain settings

**Actual Results**
A few values cannot be changed e.g. I cannot change the bed adhesion type to any value other than "skirt". When I select "brim" from the drop-down menu, the value is directly changed back to "skirt". If I change it a second time, "brim" remains selected, but the file will still be sliced as if "skirt" was selected. The same is true for other values that can also be of numeric nature. I also canot change the build plate temperature, which is now locked in at 60°C. When typing in another value, it is reset instantly back to 60°C.

**Expected results**
Settings should be possible to change

**Additional Information**
I reverted back to 20191214 where the error is not present.

Sorry for posting so many issues right now, especially since there will be a feature request following this.
Keep up the good work. Your cura build exceeds everything else that is out there right now! Thank you for the continuous work and support :)

smartavionics, 2020-01-02:
Hi, no problem with the issues but you have to remember that my builds are based on the upstream master branches and they are devils for merging badly tested changes. Anyway, as far as this problem is concerned, please attach a simple project file so that I can be sure that I have exactly the same settings when I have a look at this. No rush as I won't be looking at this until tomorrow now. BTW, happy new year!

kulfuerst, 2020-01-02:
Sure thing and happy new year to you too! The problem occurs across profiles and I cannot really make out a pattern of what settings are affected when and with which profile. The above given examples both apply to this profile however and may therefore be reproducible. 

[Cannot change values.zip](https://github.com/smartavionics/Cura/files/4017686/Cannot.change.values.zip)

smartavionics, 2020-01-03:
I have confirmed the problem but have no clue as to what is causing it so I have opened an upstream issue (https://github.com/Ultimaker/Cura/issues/6908). Hopefully they will sort it out soon.

closed 2020-01-03 by smartavionics