smartavionics/Cura #36: [FR] Support interface top layer density

kulfuerst, 2020-01-02:
This will be the last of my posts here today. I encountered the scenario when I tried to find the optimal support interface density, that a lower density support interface is much easier to break away from the printed part while not sacrificing any quality. At the same time, the strength of the support interface suffers immensely when the density is lowered anywhere beneath 85%, which hinders me from going lower as the interface then just breaks apart and can not be effectively removed from the model anymore. An option to select a separate density for the top most layer of the support interface would enable for the support interface to keep it's strength, while still being very easy to remove at low densities that would otherwise not be feasible. Thanks a ton for considering!
Best wishes

smartavionics, 2020-01-04:
Hi, I have implemented this and can make a build but it will still be based on the upstream changes that are causing a problem with values being reset. So do you want me to make a Windows build anyway for you to try or do you want to wait until the UM devs fix that other problem first?

Here's an example, it shows the two densities of support roof...

[image]

kulfuerst, 2020-01-04:
That is fantastic! Currently it is not that urgent, so you can just keep it in the pipes until you would've finished up the next release and bundle it with the rest. Great work :)

Any chance you have also fiddled around with the 0 z-distance supports I have been getting to work thanks to all your modifications? This one is the last change that will make the efficiency of that support strategy go through the roof! Other people use PVA for support with zero z-distance. I simply use your cura build :)

smartavionics, 2020-01-04:
I've reopened this issue because I want to chat about it some more.

Here's a question for you... When the top roof layer density is low (as you wish to use) and depending on the pattern used, it can leave quite big gaps around the edge of the supported area that won't get an interface line. So I am thinking of making the top roof layer always have a wall. Here's an example of what I mean, the small support has 50% roof density and 20% top layer roof density with the addition of a single wall.

[image]

As shown in this example, it can lead to some portions of the roof outline having 2 lines together (the added wall line and also the line that hugs the boundary to connect the zig/zags together.

So I guess there's actually two questions:

1 - is it useful to have the wall around the top roof layer so that the entire perimeter of that area will be supported?

2 - does the fact that in places (approx 50% of the time with zig-zag pattern and maybe more with concentric pattern) the perimeter will have two lines together cause a problem with too much support touching the model?

kulfuerst, 2020-01-04:
I think that the way that people usually support their prints and my method differ strongly enough, that the answer is pretty much the exact opposite depending on which method you'd be using.

For traditional supports with a Z-gap, an additional wall could very well be helpful in my eyes for the exact reasons you've mentioned.

With my support strategy however, it is exactly the other way around. I found what works the best BY FAR is "lines" as line type for the support roof or floor. The idea is that contact of the model to the support is limited to as few points possible while still maintaining a nice finish on the bottom. "lines" allows for this uniformly, even on the edges. You want to avoid at all costs that any lines of the models surface above the support run along with the lines of the support roof, this would create 100% contact area over a small distance, which is enough to fuse the parts together and ruin the surface finish. Lines is the optimal compromise for all desirable properties.

Why an outline would be detrimental is easiest to see when imagining printing a slope of 80° overhang from vertical. the walls of the model would make contact to parallel running lines and be impossible to release from the support. With "lines", the walls only make contact to the points on the edges. This is also why it's important to set the correct X/Y distance. I recommend 125% of the line width.

I recommend printing [this overhang test](https://www.thingiverse.com/thing:2823038) to get a good understanding of the obstacles that need to be overcome to make this work.

My inspiration came from a failed print where I had a partially clogged nozzle for only a single layer before the clog self released. Although the part was still a whole piece, I was able to cleanly break it apart at the failed layer. This partial clog is what I was trying to simulate with the settings that you so kindly provided, with a lot of success I may add :)

[0Z support.zip](https://github.com/smartavionics/Cura/files/4022583/0Z.support.zip)

smartavionics, 2020-01-05:
OK, thanks for the detailed reply. I have abandoned that idea of putting a wall around the top roof layer.

I tried printing the overhang test piece using my normal support of 75% roof density with a 0.2mm Z gap and also with the new feature of 25% density for the top roof layer and 0mm Z gap. The support came away readily in both cases and I think the part printed with the gap has less defects so I don't think I wlll be changing my support strategy yet. Still one little test piece isn't really conclusive so I will keep an open mind as to the benefits of the low density top roof layer feature.

kulfuerst, 2020-01-05:
In all honesty, I am not really surprised. My intention was to go down to 75%, maybe 70% with the new feature. The reason I couldn't do that yet is simply because the support roof adheres to the model strongly and any roof density below 85% meant that during support removal the roof would fail before the support would come loose from the model. At 25% the gaps between the lines are so large that I wouldn't expect a good surface finish anymore.
So I'm looking forward to setting the roof density to 100% and roof final layer density to 70% and seeing where that will take me. I'll post some pictures :D

smartavionics, 2020-01-06:
Understood. I'll print another piece using 100/70 as you suggest.