smartavionics/Cura #38: [FR] Implement gradient infill

extesy, 2020-01-11:
Based on the idea from https://www.cnckitchen.com/blog/gradient-infill-for-3d-prints and code from https://github.com/CNCKitchen/GradientInfill:

[image]

5axes, 2020-01-21:
The main problem with this solution : it's not possible to use it on every machine type. I have made some tests on my Ender 3 machine and the result is not correct. You need to have a direct drive feeder machine. With a bowden type extruder, it 's not possible to have sufficient precision. This is linked to the fact that the wire driving is subject to skating (if only one driving wheel) and a direct link to the stepping motor (Estep = 93). So, logically, for maximum precision of the extrusion, you need a direct drive, and a much more linear dual drive (Estep = 415) like the BMG clone, for example on a Ender 3 machine.

smartavionics, 2020-01-21:
I have a SV01 which comes with a Titan/v6 clone and it would probably work OK. But as it happens, I am not particularly keen to implement this anyway.

agibson2, 2020-02-15:
I like this.  I usually end up manually adding infill overrides at joints to make them stronger but just varying the thickness would probably work better because the infill pattern wouldn't need to change.

It would be great if this could automatically get applied only where parts of a model get smaller (walls are closer) or where parts make a sharp turn  (at joints, etc).

smartavionics, 2020-02-15:
I'm still not convinced that gradient infill is going to be much use but it's a rainy day here so I thought I would have a play. Here's a teaser screen shot...

[image]

My approach has been "cheap and cheerful" inasmuch that I am not calculating how far each infill line segment is from the boundary of the part in two dimensions (let alone 3 dimensions) because that would be very slow. So it simply use the distance along each infill line to modulate the line width. As you can see from the above image, this causes long infill lines that pass near the boundary of the object to be thinner than you may want. Whether this is a real problem in practice needs to be determined. I shall probably make a build with this tomorrow.

smartavionics, 2020-02-16:
By this evening (UK time) there should be Linux and Windows builds that have the experimental settings for gradient infill. In the end, I implemented 4 types of gradient infill. They only differ in how they determine the distance from the infill line being drawn to the boundary:

1D - trivial and fast, it simply calculates the distance to the closest end of the current infill line.

2D - bit slower, now calculates the minimum distance to the boundary on the current layer.

2D+ - same as 2D but also looks at the distance to the build plate and to the top of the print. A good choice if the model doesn't vary in height.

3D - the full monty, calculates the min distance to the nearest wall or skin (the object boundary). Could be very slow for large models and dense infill.

There's settings for specifying the min and max flows and for modifying the speed where the infill flow rate is changing.

Please experiment and report back here if you have any comments.

smartavionics, 2020-02-16:
Oh, I forgot to say that this release also has some upstream changes that may modify profiles and config files,etc. I don't know for sure but I strongly recommend backing up all your cura config files, etc. before using this release.

Personally, I put all of my cura config in a git repo and from time to time I do a snapshot so I can always get back to a working installation. I don't trust the upstream coders (much) so whenever they do anything that looks dangerous, I save my current config files so I can recover when they get trashed!

extesy, 2020-02-16:
Thanks a lot for implementing this, @smartavionics! I haven't actually printed anything yet with this feature, but I ran a few slicing experiments:
1. It only works with infill types that use straight lines but doesn't work with curved infills (concentric, gyroid, schwarz).
2. Looks like the infill line thickness is changing every 5mm or so. Can this be made configurable as well so that the transitions are a bit smoother?

smartavionics, 2020-02-16:
1 - I think it does work with gyroid and schwarz infills

[image]

2 - well, given that the extruder can't step change the line width anyway, I doubt if you will see the transitions much. Let's wait until you have done some testing before adding yet another setting.

extesy, 2020-02-16:
I've tried curved infills again and found out that they only don't work in 1D mode, probably because instead of a single line they connect lots of tiny lines so that the gradient threshold is not triggered. Works great in 2D, 2D+ and 3D modes.

smartavionics, 2020-02-16:
Ah yes, good point. I'd forgotten that.

extesy, 2020-02-16:
Hm, concentric doesn't seem to work in any mode:
[image]

smartavionics, 2020-02-16:
LOL, I didn't know that concentric infill existed!

extesy, 2020-02-16:
Yeah, I never use it myself, but I thought it's an interesting bug. 😄

smartavionics, 2020-02-16:
Just checked, the concentric infill is implemented as polygons. So are the cross and cross3d infill types. All the other infill types are implemented as a bunch of lines and they support the gradient infill. The polygons don't and I doubt if I will implement it for them.

extesy, 2020-02-16:
Fair enough, it already works for all practical use cases that I care about. Thanks again for implementing this! 👍

5axes, 2020-02-17:
Would be great also to get the possibility to change the infill printing speed as well as the flow for the people who don't have a direct drive.

smartavionics, 2020-02-17:
Try the speed factor setting. This adjusts the print speed where the extrusion rate is changing.

[image]

You can use the layer view colouring to see the speed changes.

This hasn't really been tested at all (OK, I did a couple of small prints) so it may need further development. All feedback is welcome.

5axes, 2020-02-17:
As expected the result is not so nice with just one parameter for the speed factor. If your machine doesn't have a direct drive the print speed must be inversely proportional to the material flow.
[image]
[image]

5axes, 2020-02-17:
herewith a basic test 
[image]

- Left part : Your gradient Infill ( 2D)  without speed variation, the infill width variation is hard to mesure.
- Center part : same gradient infill but with a Gradiant Infill Speed Factor = 50%
- Right part the same part previously made with the Gradiant Infill postprocessor script : and a Speed variation  50% - 200%   Flow variation 200% - 50%  [GradientInfill.py](https://github.com/CNCKitchen/GradientInfill/blob/master/GradientInfill.py)

smartavionics, 2020-02-17:
Thanks for giving it a go. I would say that the middle and right test parts are pretty similar. If anything, the middle part shows more infill width variation than the right part. Have you tried measuring the line widths with digital calipers?

5axes, 2020-02-17:
Yes, With my machine (Ender 3) without direct drive default settings width : 0.4mm  so the expected with should be for 200% 50%  -> 0.8 - 0.2 -0.8

- Left part, Gradiant Infill ( 2D) without speed variation : Average variation   0.47 - 0.38 - 0.46 
- Center part, same gradiant infill but with a Gradiant Infill Speed Factor = 50% : 0.47 - 0.73 - 026 - 0.74 - 0.5
- Right part  with the Gradiant Infill postprocessor script :  0.7 - 0.24 - 0.71

hereafter the sample code from the script
[GradientMaster.zip](https://github.com/smartavionics/Cura/files/4216700/GradientMaster.zip)
 - 0.71

smartavionics, 2020-02-18:
Thanks for the measurements.

> Center part, same gradiant infill but with a Gradiant Infill Speed Factor = 50% : 0.47 - 0.73 - 0.26 - 0.74 - 0.5

Don't understand what the first and last values are.

Anyway, my next release will provide a new setting (gradient infill speed scheme) which will let you choose between what it does now (speed is modified where the infill flow is changing) and the scheme you are asking for (constant extrusion rate). Constant extrusion rate is fab as it will produce nicely modulated line widths and generally speeds up the prints for infills that have straight lines. It would probably be death for infills like gyroid as the printer would be in danger of flying apart on the fast sections.

5axes, 2020-02-18:
Without a direct drive, The speed reduction must be (in my case) proportionnal to the Flow variation.  In my test the width of the line closed to the border is smaller than the width at the beginning of the flow increase . That's why   I have these values.

[image]

[image]

smartavionics, 2020-02-18:
I still don't understand why a bowden extruder should behave as you are saying. What I think is happening in your case is simply that the extruder/hotend cannot deliver 200% at the normal infill speed. It's nothing to do with the fact that it's a bowden extruder.

5axes, 2020-02-18:
Yes you're right the main issue comes from the limitation of this type of extruder to deliver 200% at the normal infill speed, but it's not because it's a bowden extruder. It's because, on my machine with the original extruder, it is not possible to have sufficient precision. This is related to the fact of the extruder is subject to skating (only one driving wheel) and to a direct link to the stepping motor (Estep = 93). So, logically, for a maximum precision of the extrusion, you need a direct drive, and a much more linear dual drive (Estep = 415) as the BMG clone for example on the Ender 3/5.

smartavionics, 2020-02-18:
At the end of the day, it's probably a combination of factors that stop the extruder/hotend performing as well as we would like.

Anyway, I am going to make a new build this afternoon that has the choice of speed modification algorithms and, hopefully, that will give you what you want. Stay tuned...

smartavionics, 2020-02-18:
New release now on dropbox 20200218, please give it a go.

5axes, 2020-02-18:
As someone as already write : Faster than the light !

5axes, 2020-02-18:
I think I need a little bit more explanations .

- None : Ok it's clear the Infill speed is used everywhere 

- On Flow Change : It's the previous behavior , Speed reduction around the Gradient Infill distance area.

- Fast ? -> Inside the gradiant Infill area the speed gradualy change from the Infill speed to a value equal to { Infill Speed * Gradient Infill Speed Factor }  Outside the gradient area Speed =  { Infill Speed * Gradient Infill Speed Factor } Ie for 100mm/s  100% = 200 mm/s ?

- Slow ? Outside the gradient area Speed =   Infill Speed . In the gradient fill area the Speed is reduce from  Infill Speed  to 100%  = minimum speed Ie : 100mm/s -> 10 mm/s ?

[image]

smartavionics, 2020-02-18:
The fast and slow schemes both modify the print speed so that throughout all of the infill, the extruder will turn at the same speed. So for both schemes, when the flow rate doubles, the speed has to halve, and so on.

The fast scheme uses the normal print speed for normal flow so if that normal infill speed is 100 mm/S you will indeed get 200 mm/S where the flow is 50% and 50 mm/S where the flow is 200%. This is assuming the infill speed factor is 100%. If you modify that the speed changes will be less or more depending on whether you reduce the speed factor or increase it. Note that changing the speed factor from 100% will make the extruder rate change.

The slow scheme uses the normal print speed for the lowest flow and so all other flows will make the print speed reduce. So in your example, with the slow scheme, at 50% flow the speed will be 100 mm/S, at 100% flow the speed will be 50 mm/S and at 200% flow, the speed will be 25 mm/S.

That's the intention.

kulfuerst, 2020-03-18:
Hey everyone. I just cannot find this script! Is there something I need to do to enable it? I am running 20200306. The post script selection looks like this for me. Has gradient infill been removed again?
[image]

smartavionics, 2020-03-19:
My implementation of gradient infill does not use a script. It is controlled by settings in the experimental section.

[image]

kulfuerst, 2020-03-21:
Thanks for your help. I've found the settings and, oh man, is this epic! Whether this actually helps with part strength or not, it looks super cool and is just awesome to be able to do. Absolutely fantastic job implementing this!
Someone tell Stephan from CNC Kitchen about this

smartavionics, 2020-03-21:
Good. Please let me know if you find any problems as I don't actually use this feature myself (yet).

dyordan1, 2020-04-22:
Just watched the cnc kitchen video and it led me down to this - did you consider stepping back and trying to solve the posed problem, rather than implementing Stephan's hack he could think off in the moment?

Specifically, did you try exploring implementing a gradient over the pattern density itself, rather than the outflow rate? Should make this very useful infill strategy compatible with any printer and eliminate much of the speed complexity. Was thinking of doing the same, but have not contributed to slicer software before. Could be fun :)

smartavionics, 2020-04-22:
No, I haven't considered how the infill density could be modulated by the distance to the walls. Thinking about it now, the discontinuities that would be present where the infill density changes would (a) make the infill weaker and (b) make it slow to print. Sounds like a lot of work for dubious gain.

5axes, 2020-04-22:
We already have in Cura such type of Infill Pattern It's call Cubic Subdivision . But on my point of view it's not very efficient.

smartavionics, 2020-04-22:
I suspect that @dyordan1 is talking about varying the infill density within a single layer depending on the proximity of the nearest wall.

5axes, 2020-04-22:
Yes I think so too, but even if in the Cubic Subdivision we dont' have a notion of "Distance" it's closed to a variation of the infill density. But I don't know as it is a mathematical subdivision by 8 of each cube if it would be possible to limit this subdivision in an area limited by  a distance parameter.

dyordan1, 2020-04-22:
Yeah, I guess I was talking about varying density within each layer based on wall proximity. As you pointed out, @smartavionics, I'm sure the discontinuities would be tough to handle, especially with certain types of infill. With that said - I do think it warrants an experiment. My gut sense is you might get OK results with rectilinear fill - the discontinuities would mesh just fine, especially if you're smart about density steps. Then, maybe ask Stephan to do some of his linear/load strength tests with the new feature and compare the results.

To Stephan's point in the original video, the load on a part falls off from the skin onwards, i.e. it would follow the density curve. It would make sense the optimal point for strength vs material used will add material that roughly approximates load pattern. I kinda like Cubic subdivision but I think the rough cube shapes it ends up using are not sufficient to distribute load in a complex shape. There are also rather large holes in the infill, which puts extra load on the cube walls, in combination with sharp corners which don't distribute load too well. Varying density would not cause as big of a gap, since even the sparser parts could have some infill and could be a more complex shape that distributes load better.

Btw, none of this intends to devalue your awesome work here, I'm just suggesting an alternate look at it - if I find the time to do so anytime soon, I would love to experiment myself but I'm sure I'd fumble and fall much more than one of y'all :)

smartavionics, 2020-04-22:
I agree that infill could be better optimised to take into account not only the proximity of walls but also the load bearing requirement of a given infill region. Currently, where I need denser infill, I use modifier meshes to alter the infill pattern and density where required, here's an example that puts dense infill around the holes for fasteners on a bracket and uses sparser gyroid infill elsewhere...

[image]

MasterCATZ, 2021-04-19:
was this removed ? can not find it in beta 4.9

smartavionics, 2021-04-19:
It's still in my releases. It's not in the UM releases.

FabioM3, 2023-09-27:
A thing that need to be taken in account is the 3D of the shape, the infill needs to be more thick at the borders, and less in the middle.

dbaarda, 2025-09-01:
I have a few quick comments/observations;

1. The density calc is only done based on the distance to the sides for the currently printing D2 layer slice, not the distance to the closest surface face in 3D space. This means it doesn't take into account how close horizontal faces above/below are, so the density doesn't increase just below top or just above bottom surfaces. Of course this makes it relatively cheap and easy to calculate, and calculating 3D distances probably requires deeper integration to analyse the model than just a post-processing script (as is done for Adaptive Cubic/Cubic Subdivision).
2. It only adjusts the line thickness of the infill pattern lines, which limits the strength and range of density adjustment. As mentioned above in #issuecomment-617530346 in theory it should be possible to vary the infill density. For many infill patterns this would cause discontinuity problems as mentioned in #issuecomment-617581992, but for equation-based infills like gyroid and TPMS it is possible to continuously vary the density without discontinuitys. The marching squares algorithm for calculating these implemented in https://github.com/SoftFever/OrcaSlicer/pull/10360#issue-3306456744 can easily generate infill patterns for any arbitrary equation, so could easily handle variable density infill patterns.