smartavionics/Cura #40: Fill Gaps Between Walls: wall and skin not actually filling tiny features.

JohnEdwa, 2020-01-29:
Slicing this tiny cable clip where there is no space for a full wall, only "Everywhere" actually fills the gap successfully even though according to the preview the fill is done as a shell top/bottom layer, and as I'm asking for 3 walls it should be a thin inner wall, not skin (so it should be filled even with "Only Wall Gaps" setting).

Project file: [CE3_Cable_Clip.zip](https://github.com/smartavionics/Cura/files/4129399/CE3_Cable_Clip.zip)

[image]
[image]

**Application Version**
20200125

Also, a side note, the explanation help popup for this quite confusing. My suggestion would be to completely re-order and reword it to something such as:

"Fills the gaps between walls where no walls fit. 
Only Wall Gaps fills the gaps between adjacent wall lines.
Wall and Skin Gaps fills the gap between walls, and between a wall and the skin.
Everywhere fills the gaps between walls, a wall and the skin, and a wall and the infill."

(Although I'm a non-native English speaker/writer so I'm probably not the best for this)

smartavionics, 2020-01-29:
Yes, it's not working right. I will investigate, thanks for the report.

smartavionics, 2020-01-30:
I have removed the wall and skin gaps option as it wasn't working right and I can't be bothered to fix it. So the next release will have options: nowhere, only wall gaps and everywhere.

closed 2020-01-30 by smartavionics

JohnEdwa, 2020-01-30:
Not ideal, but understandable. I'll continue using everywhere, as the main reason I want it enabled is for filling visible gaps on the top layers which this impementation of "walls only" will leave unfilled.

5axes, 2020-01-30:
Thank's for your contribution @smartavionics , this is with this type of example that I know why I 'using your release rather than the official.
[image]

smartavionics, 2020-01-30:
You're welcome.

What I really don't understand is why UM do not fix some of the long standing issues that Cura suffers from. Sure, they have to put most of their effort into stuff that is relevant to UM printers as that's how they get paid but a lot of the worst bugs/features in Cura affect all printer types. So this particular issue (poor quality gap filling and thin line printing) along with Cura's poor handling of walls whose width is not close to an even number of line width should have been fixed years ago but they don't seem to care.

noisettetbou, 2020-02-02:
Hi. I use curamaster 20200125 and the option "fill gaps between walls" works. I think you've got a little bug but the idea is well and I hope this option is back in a future release.

smartavionics, 2020-02-03:
I just removed the Wall And Skin Gaps option, the other options are still there.