smartavionics/Cura #41: Font glitches, shifted rendering

LightTemplar, 2020-01-30:
**Application Version**
Cura-mb-master-x86_64-20200125

**Platform**
OS: KDE Neon
KDE Plasma: 5.66.0
QT: 5.13.2
Kernel: 5.3.0-28-generic
Two Videocards, Intel i915 and Nvidia GTX 1050, 
Driver: nvidia-driver-440, ver: 440.48.02
Second Monitor via Lenovo Dock, using DisplayLink drivers.

**Information**
Right after start it in shows shifted text overall in UI, except of main menu.

I am using bigger fonts, with DPI=106, but resetting it to 96 didn't help much.

Starting in terminal shows following:
```
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
QObject::connect: Cannot queue arguments of type 'QList<QQmlError>'
(Make sure 'QList<QQmlError>' is registered using qRegisterMetaType().)
Model size of -76 is less than 0
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
Model size of -77 is less than 0
```
[image]

smartavionics, 2020-01-30:
Hi. The terminal output looks normal so there's nothing useful there.

Do you get the same problem irrespective of which monitor the window opens on?

LightTemplar, 2020-01-30:
Yes, both monitors show the same problem.

smartavionics, 2020-01-31:
Did you take a look at https://doc.qt.io/qt-5/highdpi.html ? Is it worth setting QT_AUTO_SCREEN_SCALE_FACTOR as they suggest?

LightTemplar, 2020-01-31:
You mean starting it like that: `QT_AUTO_SCREEN_SCALE_FACTOR=1 ./Cura-mb-master-x86_64-20200125.AppImage`? It didn't change anything. On both monitors.

smartavionics, 2020-02-01:
Hmm, I'm running out of ideas. Are you using any plugins that alter the screen layout at all?

LightTemplar, 2020-02-04:
At least I have localized it - I've created a fresh new user, tested the appimage there and it looked nice! Just how it should. 

I use kvantum, but I don't know if it works - for now I've removed it from all settings, where I could find it.

By the way, thank you for your algorithm, right filling the space between walls - with it my last printing was about 30% faster. I've printed this https://www.thingiverse.com/thing:3078296

smartavionics, 2020-02-04:
I know nothing about kvantum but I just looked at the installation instructions for that and I wonder, have you tried clearing QT_STYLE_OVERRIDE before starting cura?

LightTemplar, 2020-02-04:
It's empty in my system. But I tried to set different values. With "Breeze" it said  
```
QApplication: invalid style override passed, ignoring it.
    Available styles: Windows, Fusion
```
So I tried "Windows" and "Fusion" - the output again had those glitches . 

Next I will try to purge kvantum and tell you results.

LightTemplar, 2020-02-16:
Purging Kvantum didn't help either.