smartavionics/Cura #42: v20200208 - not slicing

amagr0, 2020-02-13:
Application Version - CURA Master - V20190724
Platform - Win 10 Pro, 64 bit
Display Driver - NVIDIA GeForce 840M

Hello @smartavionics,

The latest version is not slicing, it starts, then slicing bar stops in the middle forever.
This stl with other version of Cura (not the previous) was sliced in less that 10 seconds.

[image]

Thanks.

smartavionics, 2020-02-13:
Can you please provide a project file that I can test. Thanks.

amagr0, 2020-02-13:
[v20200208_not_slicing.zip](https://github.com/smartavionics/Cura/files/4197968/v20200208_not_slicing.zip)

smartavionics, 2020-02-13:
Thanks. It slices fine on Linux. I'll fire up a Windows machine and try that.

smartavionics, 2020-02-13:
It also slices fine on my Windows 10 machine.

Can you provide the cura.log and also the stderr.log files?

amagr0, 2020-02-13:
[v20200208_not_slicing_logs.zip](https://github.com/smartavionics/Cura/files/4198061/v20200208_not_slicing_logs.zip)

smartavionics, 2020-02-13:
It failed because it's missing a setting called bridge wall end boost. If you enable the bridge settings and set the expert setting visibility, you should see it like this here..

[image]

It's visible on my Windows laptop, is it not visible for you?

amagr0, 2020-02-13:
I don't have that option. All the parameters are visible.

[image]

smartavionics, 2020-02-13:
That's very odd because the setting is definitely in the release. Please remove the installation by completely deleting \program files\ultimaker cura master (or wherever it is installed) and try reinstalling. It looks like the file that contains the settings (resources\definitions\fdmprinter.def.json) has not been updated.

amagr0, 2020-02-13:
I already uninstalled and deleted the Master files, but the problem remains (actually now I have other problems - configuration errors).

[image]

amagr0, 2020-02-13:
I found a solution by deleted the cache folder in \AppData\Local\cura\master

[image]

closed 2020-02-13 by amagr0