smartavionics/Cura #43: Bridge issue

eskeyaar, 2020-02-14:
**Application Version**

win64-20190912

**Platform**

Win64
**Printer**

Attached 3mf

[LM_Bridging_Test_50.zip](https://github.com/smartavionics/Cura/files/4205517/LM_Bridging_Test_50.zip)

**Steps to Reproduce**

Default Bridge settings
**Actual Results**

Some layers it starts from middle, resulting in not attaching to it.  
[image]

**Expected results**

**Additional Information**

Printed same model in Prusa mini, using prusa slicer, it came out good.

smartavionics, 2020-02-14:
Hi. Unfortunately, I can't load your project because it uses an unknown printer. I sliced it for another printer but can't see the same problem occurring. Tell me, what layer number are you showing in the image above?

eskeyaar, 2020-02-14:
Hi, in my current project i get those things on layer 10 , 38 ,54. Since my machine is lulzbot mini, i am using line width of 0.6. 
changing the line width to 0.6 wall line count to 3, on ultimaker 2+ on normal stock profile, can generate this issue too for your reference.

smartavionics, 2020-02-14:
Ah, I understand now. What's happening is that because of the very wide wall lines, there isn't enough space to fit all the walls completely. Some walls get "broken" and they don't extend all the way across the bridge region. Hmm, that's a tricky one. I will think about what to do in this situation.

Could you please attach the gcode produced by the prusa slicer so I can see how they have handled this. Thanks.

eskeyaar, 2020-02-14:
Great. 
Attached Gcode from PrusaSlicer.
[Bridging_Test_50_0.2mm_PLA_MINI_57m.zip](https://github.com/smartavionics/Cura/files/4206314/Bridging_Test_50_0.2mm_PLA_MINI_57m.zip)

Thanks

smartavionics, 2020-02-15:
Thanks for the prusa gcode. Looking at that, I can see that they are doing nothing special. They simple manage to fit the 6 wall lines into each of the spokes. Unfortunately, due to the way that Cura handles walls, it can't do the same. It's not really a problem specific to bridging, it's just how Cura fits wall lines into the available space. Sorry, I can't make Cura do a better job when using 0.6mm wide lines on this model. I did print it using 0.4mm wide lines and it came out very well indeed with not a single dropped bridge line.

eskeyaar, 2020-02-17:
Thank you Mark for your time to look at it. 
I will try to change my wall lines count and see. Do you want S3D gcode version to see also? 
Thanks.

smartavionics, 2020-02-17:
No, thanks, I don't need to see the S3D code.

I'll close this issue now as there's nothing I can do to solve the problem which is fundamental to how Cura generates walls.

closed 2020-02-17 by smartavionics