smartavionics/Cura #44: Menu text obscured in Raspberry Pi AppImage Cura-mb-master-armhf-20200208

n2htt, 2020-02-16:
**Application Version**
Cura-mb-master-armhf-20200208.AppImage

**Platform**
Raspian Buster

**Printer**
n/a

**Steps to Reproduce**
Execute the appimage - UI comes up with all text blocks in menu and dialogs as black boxes.

**Actual Results**
UI is not usable.

**Expected results**
Dialog and menu text should be readable.

**Additional Information**
Cura-mb-master-armhf-20200201.AppImage works fine on the same machine.

Thanks
Mike N2HTT

smartavionics, 2020-02-16:
Hi Mike,

Sorry that the last build isn't working for you. Is your Pi up to date ? Have you done "sudo apt-get update" and "sudo apt-get upgrade" recently?

If your Pi is up to date and it still doesn't work, could you please do the following...

Run the the AppImage and in a terminal do "ps aux | grep cura" to find the PID of the cura process. Then do "lsof -p PID" to list all the open files that cura has and post that list here. (I don't think it contains any secrets)

n2htt, 2020-02-16:
look like everything was up to date, here is the console output:

[issue 44 console.txt](https://github.com/smartavionics/Cura/files/4209854/issue.44.console.txt)

BTW the prior version works beautifully on my pi, seems a bit faster than cura running on my old window box.

Thanks
Mike

smartavionics, 2020-02-16:
Thanks for the console text file, I will compare to what I have here. If I can't solve this, I will revert back to using desktop GL for the pi build.

smartavionics, 2020-02-16:
Could you try removing the contents of /home/pi/.cache/cura/qmlcache  ? Thanks.

smartavionics, 2020-02-16:
Also, do you have /usr/lib/arm-linux-gnueabihf/libGLESv2.so ? What does ls -l show for that file?

n2htt, 2020-02-16:
I tried clearing the cache, but that did not help.
The file does not appear to be installed on my pi

pi@raspberrypi:~ $ ls -l /usr/lib/arm-linux-gnueabihf/libGLESv2.so
ls: cannot access '/usr/lib/arm-linux-gnueabihf/libGLESv2.so': No such file or directory
pi@raspberrypi:~ $

smartavionics, 2020-02-16:
Ah, that's the problem. I'll find out where that comes from, standby...

smartavionics, 2020-02-16:
It's in a package called libgles2. Please install that and try again.

n2htt, 2020-02-16:
Although the library doesn't show up in the directory you suggested, apt indicated that libgles2 was already installed. Just to be sure, I removed and re-installed the package, but no change of behavour. 

Here is the console with the apt interactions
[console2.txt](https://github.com/smartavionics/Cura/files/4210143/console2.txt)

smartavionics, 2020-02-16:
So this is what I see...

[image]

n2htt, 2020-02-16:
Problem solved!
I was not seeing the files because I didn't **sudo ls**.

Once I could see them, I noticed that you had two soft links that I do not have. Added the soft links and now 0208 is working fine.

pi@raspberrypi:~ $ sudo ls -l /usr/lib/arm-linux-gnueabihf/libGLES*
lrwxrwxrwx 1 root root    21 Jan 22 07:04 /usr/lib/arm-linux-gnueabihf/libGLESv1_CM.so.1 -> libGLESv1_CM.so.1.2.0
-rw-r--r-- 1 root root 25992 Jan 22 07:04 /usr/lib/arm-linux-gnueabihf/libGLESv1_CM.so.1.2.0
lrwxrwxrwx 1 root root    18 Jan 22 07:04 /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2 -> libGLESv2.so.2.1.0
-rw-r--r-- 1 root root 58760 Jan 22 07:04 /usr/lib/arm-linux-gnueabihf/libGLESv2.so.2.1.0
pi@raspberrypi:~ $ cd /usr/lib/arm-linux-gnueabihf/
**pi@raspberrypi:/usr/lib/arm-linux-gnueabihf $ sudo ln -s libGLESv1_CM.so.1 libGLESv1_CM.so
pi@raspberrypi:/usr/lib/arm-linux-gnueabihf $ sudo ln -s libGLESv2.so.2 libGLESv2.so**
pi@raspberrypi:/usr/lib/arm-linux-gnueabihf $ ls -l libGLES*
lrwxrwxrwx 1 root root    17 Feb 16 11:53 libGLESv1_CM.so -> libGLESv1_CM.so.1
lrwxrwxrwx 1 root root    21 Jan 22 07:04 libGLESv1_CM.so.1 -> libGLESv1_CM.so.1.2.0
-rw-r--r-- 1 root root 25992 Jan 22 07:04 libGLESv1_CM.so.1.2.0
lrwxrwxrwx 1 root root    14 Feb 16 11:54 libGLESv2.so -> libGLESv2.so.2
lrwxrwxrwx 1 root root    18 Jan 22 07:04 libGLESv2.so.2 -> libGLESv2.so.2.1.0
-rw-r--r-- 1 root root 58760 Jan 22 07:04 libGLESv2.so.2.1.0
pi@raspberrypi:/usr/lib/arm-linux-gnueabihf $ 
I don't know why the soft links were not there but that was the issue.

Thanks for the assist
Mike

smartavionics, 2020-02-16:
Hmm, weird. I don't remember creating those symlinks manually. Oh, well, glad it's working for you.

I'll close this issue now. If anyone else has a problem I will put something into the README.

closed 2020-02-16 by smartavionics

jpfnos, 2020-02-26:
I had the same problem with version Cura-mb-master-armhf-20200218.AppImage.

Following these steps, I was able to correct the problem:

> Problem solved!
> I was not seeing the files because I didn't **sudo ls**.
>
> Once I could see them, I noticed that you had two soft links that I do not have. Added the soft links and now 0208 is [...]