smartavionics/Cura #46: Just to say thank you ! This 32 bit release is a great Idea

5axes, 2020-02-28:
I haven't test it for the moment. But It's going to help a lot of people who still had an old computer and who was stuck with their version 1.15 of Cura.

smartavionics, 2020-02-28:
Thanks, I made it as an experiment to try and help with the Ryzen crashes but it turns out even the 32 bit build has the same issue.

If you get around to trying it on a real 32 bit machine, please let me know if it installs and runs OK.

I will continue to make both 32 bit and 64 bit builds from now on.

5axes, 2020-03-02:
Just installed on a Win7 32 bit config  , a little bit long to start (it' a 10 year old computer) but seem's to work fine.  Your version will certainly help a lot of people.

smartavionics, 2020-03-02:
Great, thanks for letting me know. I will continue to make 32 bit releases. Cheers!

closed 2020-03-02 by smartavionics