smartavionics/Cura #47: Issue with mesh modifier settings ("Mesh type: Modify settings for overlaps")

kvnper, 2020-03-01:
First of all, thank you for your great work with your personal releases of Cura!

Using version Cura-mb-master-win64-20200226, I have run into an issue on every model I've tried so far where I cannot change the 'Per model' settings for any mesh when setting the mesh type to "Modify settings for overlap" (image below for reference) Only these two options are shown for reasons unknown (maybe from a previous saved state).

[image]

As you can see here in this screenshot below, I have ticked various settings but the options do not update on the left black box.

[image]

Then clicking off all models and selecting the mesh again removes any options:

[image]

Ticking and unticking "Infill only" brings back those two options from earlier.
[image]

Rinse and repeat from the beginning.

I can confirm this works correctly in the latest Cura 4,5

smartavionics, 2020-03-01:
Hi, yes, it's totally broken, isn't it? Well, that's the work of the upstream UM developers, not me. I have commented on the changes they made and hopefully they will un-break it. I will see if I can make a temporary fix in the meantime.

smartavionics, 2020-03-01:
Yes, I reverted the changes they made and it appears to work OK so I will build a release today.

smartavionics, 2020-03-01:
0301 release now available, thanks for the bug report.

closed 2020-03-01 by smartavionics