smartavionics/Cura #48: [Cura-mb-master-win32-20200226] Modify settings for overlap

5axes, 2020-03-02:
Don't know if it's an issue with just the Win32 release. I must make some test with the win64 release.  But it's not possible to select some  particular settings for an object. Hereafter the error message in the log file concerni ng this issue.
`
2020-03-02 18:10:06,556 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/plugins/PerObjectSettingsTool/PerObjectSettingsPanel.qml:209:29: Unable to assign [undefined] to qulonglong
2020-03-02 18:11:43,468 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingTextField.qml:155: TypeError: Type error
2020-03-02 18:11:43,507 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingTextField.qml:158: TypeError: Type error
2020-03-02 18:11:43,546 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingTextField.qml:99: TypeError: Type error
2020-03-02 18:11:43,585 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingItem.qml:173: TypeError: Type error
2020-03-02 18:11:43,625 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingItem.qml:144: TypeError: Type error
2020-03-02 18:11:43,663 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingItem.qml:150: TypeError: Type error
2020-03-02 18:11:43,702 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [399]: file:///C:/Program Files/Ultimaker Cura master/resources/qml/Settings/SettingItem.qml:31: TypeError: Type error
`

5axes, 2020-03-02:
Tested on version mb-master-20200218 Win 64 Works Fine

smartavionics, 2020-03-02:
Which settings could you not select?

5axes, 2020-03-02:
In this case every settings relativ to the Infill parameters : pattern or infill percentage
But I have made some tests with the other settings and it's the same situation.

smartavionics, 2020-03-02:
OK, thanks. I just tried my other 32 bit build which is for the PI 4 and that doesn't appear to have a problem. I will investigate.

5axes, 2020-03-02:
I'm currently downloading the **_Cura-mb-master-win32-20200301_** release. I will try also with this release.

smartavionics, 2020-03-02:
Note this line from README.md...

* 20200301 - Fixed upstream breakage of per-object settings + fixed support block picking on armhf (PI 4) build.

5axes, 2020-03-02:
Ok thanks 20200301   issue solved !

closed 2020-03-02 by 5axes