smartavionics/Cura #49: Fan lag not working? Also crashing?

MastaRob, 2020-03-02:
Thanks for working on the build! Posting here after emailing you, found another issue. The "fan cooling lag" - I assumed this was supposed to turn on the fan, at the user specified time before it is needed? Unfortunately this is not working for me. After a dodgy print I checked with a very simple test file:

[image]

... and then looked over the GCode, this is the section where the fan turns on:

;MESH:NONMESH
G0 F2400 X168.431 Y148.047 Z1.09
G0 F12000 X167.841 Y151.451
;TIME_ELAPSED:132.553110
;LAYER:5
;TYPE:WALL-OUTER
;MESH:test.stl
M106 S255
G1 F960 X167.454 Y152.181 E0.03679

So almost immediately after the Z height is increased. I changed the fan lag back to 0 and it made no difference to the gcode.

Second issue is crashing. After I leave it running in the background for a while. Possibly related to putting PC to sleep. When I come back and try to use again, cura sometimes crashes - windows log gives me the below report. Only difference from stock is I havethe duetRRF plugin. Running windows 7. Still investigating if normal cura 4.5 crashes, will report when I am more certain. So far I've had one crash on this version and nothing on stock 4.5 but will keep them both open and see what happens.

Anyway, thanks for all your work on this. 

Faulting application name: Cura.exe, version: 0.0.0.0, time stamp: 
0x5e53c212
Faulting module name: ntdll.dll, version: 6.1.7601.24384, time stamp: 
0x5c6e245d
Exception code: 0xc0000374
Fault offset: 0x00000000000bf302
Faulting process id: 0x4acc
Faulting application start time: 0x01d5efba101f9614
Faulting application path: C:\Program Files\Ultimaker Cura
master\Cura.exe Faulting module path: C:\Windows\SYSTEM32\ntdll.dll
Report Id: 2b664ad2-5c74-11ea-9d9a-6c626db629a2

smartavionics, 2020-03-02:
The fan lag setting is there to tell Cura how long it takes the fan to spool up from stationary to full speed. With that knowledge, it can turn the fan on/off before the airflow change is required. i.e. when printing a bridge wall line that requires 100% fan, it can turn the fan on full, say, 1 second before the nozzle gets to the bridge wall line. That's the idea, anyway.

As for the crash, I can't do anything about that because I don't use Windows and have no knowledge/tools to help debug crashing Windows programs. Sorry.

MastaRob, 2020-03-02:
No problem with Windows. OK yes, thats what I thought fan lag was supposed to do, switch on before it is needed. Please let me know if you can reproduce, and if a fix then happens - unfortunately it looks like I cannot use the overhang fan speed without the feature, which is a shame because it's a great feature.

smartavionics, 2020-03-02:
Perhaps you could save the project file, zip and attach to this issue so I can investigate. TBH, I don't really follow what you're saying.

MastaRob, 2020-03-02:
OK let me try again - fan speed lag isnt changing when the fan turns on. If I set regular fan speed 30%, fan speed lag 1.5s, overhang fan speed 100%, then the fan stays at 30% right up until the start of the overhang print. And then it spools up while printing the overhang, meaning the start of the overhang is a mess.

smartavionics, 2020-03-02:
Ah, OK, there's a limitation to the spooling up thing which is that it only works within individual layers. So if at the end of one layer the fan is, say, 30% and the next layer immediately requires 100%, the fan won't start spooling up until the start of the layer.

But why don't you just set the fan speed to 100% from layer 1 upwards?

MastaRob, 2020-03-02:
Blast! OK hmm, maybe I can change print order.
Having the fan on reduces inter-layer adhesion, and leaves the model weaker. So its actually counter-productive to run the fan in areas where it isnt needed.

MastaRob, 2020-03-08:
Using newest version from the other day, I dont think fan speed lag is working at all - watching a print going through, and fan keeps flicking up to max for just a split second and then back again. All on same layer. Looking over the gcode confirms this, with sections like:

G1 X43.074 Y46.339 E0.02388
M106 S255
G1 F812.8 X42.57 Y46.304 E0.0189
M106 S76.5

smartavionics, 2020-03-09:
Please provide a project file, so I can check the settings.