smartavionics/Cura #50: problem of quality

landrieve, 2020-03-06:
Hi
i have a problem with the quality betwen Cura3.6 and cura master4.5 on a xyz calibration cube. I dont understand what happen (the mark upper and under the X) and why this difference. i use the same profil and the same gcode start. I use a ender3. sorry for my english;
[image]

[gcode.zip](https://github.com/smartavionics/Cura/files/4297126/gcode.zip)

smartavionics, 2020-03-06:
Hi, the file CE3_xyzCalibration_cube cura3_6.gcode doesn't seem to contain gcode. Do you have another copy?

smartavionics, 2020-03-06:
It looks like the contents of a file system directory in binary format!

smartavionics, 2020-03-06:
I think the problem is likely to be the unretracted travel moves that occur on the layers at the top and bottom of the letters. Try setting the max comb distance with no retract to something small (e.g. 10).

landrieve, 2020-03-06:
sorry for the gcode; I think this one is good.
[gcode 3_6.zip](https://github.com/smartavionics/Cura/files/4297269/gcode.3_6.zip)

landrieve, 2020-03-06:
on cura 3.6 the max com distance is 0. i try tne print of xyz calibration cube with max comb distance =0 on cura 4.5.

smartavionics, 2020-03-06:
No, it has to be > 0, try 10.

landrieve, 2020-03-06:
ok, i have trying with 0 and i have the same thing that with 30. i try with 10

landrieve, 2020-03-06:
i have trying with 10 and i have the same thing that with 30. i'm sorry but the best result for me is with the 3.6 of Cura. thank's for your help

smartavionics, 2020-03-06:
Sorry, must be some other setting that is causing the issue.

closed 2020-03-06 by smartavionics