smartavionics/Cura #51: Slicer stalls when bridge settings enabled

kvnper, 2020-03-11:
Hi SA,

I ran into an issue slicing a model when 'Enable Bridge Settings' is enabled. When enabled, the slicing stalls and does not proceed past half way. When disabled it slices swiftly. I haven't tried leaving it more than 10 minutes but usually, in regular cura, this setting does not increase slice time and paired with the fact that you modified the bridging code, I thought that it might be an issue with that.

I've attached the 3mf file (had to rename the extension to attach it here).

[CFFFP_octofinger (1).3mf.txt](https://github.com/smartavionics/Cura/files/4317495/CFFFP_octofinger.1.3mf.txt)

kvnper, 2020-03-11:
I forgot to mention that it slices fine in cura 4.5 with the setting both enabled and disabled.

smartavionics, 2020-03-11:
With my latest build it doesn't stall. Can you please provide the stderr.log and cura.log files for when it fails for you?

[image]

There's something I don't understand though which is the fact that the cross pieces are not being treated as bridges so I need to investigate why that is occurring.

kvnper, 2020-03-11:
Ok I thought I had the latest build, but I'll try installing it again.

I'll get back to you with the logs shortly if I still have the same issue.

kvnper, 2020-03-11:
No luck with the latest build 20200306

Log attached. I can see an error there related to 'bridge_extend_floating_walls'

[cura.log](https://github.com/smartavionics/Cura/files/4317784/cura.log)

kvnper, 2020-03-11:
I should really get some sleep, I keep missing things! stderr.log attached.

[stderr.log](https://github.com/smartavionics/Cura/files/4317789/stderr.log)

smartavionics, 2020-03-11:
I think that can be cured bu removing the cura cache files. On Linux they are to be found at ~/.cache/cura/master/. I don't know about Windows and I don't have a Windows machine running at the moment but I guess they would be under AppData ? AppData/Roaming?

kvnper, 2020-03-11:
I deleted everything in that folder:
[image]

Presumably it gathered the settings from my 4.5 install. The thing is this folder stores the settings for custom printers, deleting anything here deletes any custom printers, print profiles, configurations, and settings.

Even so, I backed up and deleted every install in /Roaming/cura and I'm still having the slicing issue.

I notice I'm not getting the "Bridge Extend Floating Walls" setting, even after trying to show all settings.

[image]

kvnper, 2020-03-11:
Update: I found a cache in /AppData/Local/cura, I backed up and deleted all folders inside. The issue seems to be fixed! Sliced correctly and the setting is visible. Thanks for your help!

smartavionics, 2020-03-11:
Glad you fixed the hangup.

I discovered that in your profile you have set the small hole max size rather large and that overrides the bridge wall detection. I reduced the value and the bridge walls are printed as expected.

closed 2020-03-11 by smartavionics