smartavionics/Cura #52: [FR] Make bridges use mm³/mm instead of flow

kulfuerst, 2020-04-02:
**Application Version**
20200306

**Platform**
Win10

**Expected results**
Tune your flow once and never worry about it again.

**Actual Results**
Flow affects the extrusion during a bridge in relation to the layer height. This means that at low layer heights, almost no material is extruded and bridging performance is poor, while at high layer heights too much material is extruded, the bridges perform bad as well.

**Additional Information**
If the flow during a bridge was calculated in mm³/mm, the bridging behavior would only need to be calibrated once and bridges would forever print perfectly, no matter the layer height.

I've got no clue if this is actually doable but it would be absolutely awesome!

kulfuerst, 2020-04-15:
I've just now realized that mm³/mm is the weirdest unit. It's supposed to mean mm³ extruded per mm printed.

x0rtrunks, 2020-04-28:
I assume you mean the volumetric flow rate?

I too wish Cura was more volumetric flow rate aware.

kulfuerst, 2020-04-29:
@x0rtrunks volumetric flow rate sounds like it describes what I had in mind. Yeah, let's go with that!