smartavionics/Cura #53: randome print stops

badwolf1972, 2020-04-19:
Hi good people anyone else having random print stops ...... i am a lot of them
my setup
Raspberry pi 4 4GB
rasbian os up to date
Cura-mb-master-armhf-20200328
Geeetech A10 

starts printing with no problems but randomly stops no errors on printer or raspberry pi 

and ideas anyone??

smartavionics, 2020-04-19:
Hi, sorry to hear you are having problems. I have never tried printing from Cura using USB (I guess that is what you are doing). Could you please zip up the ~/.local/share/cura/master/cura.log and attach to this issue so I can see if there are any errors there. Thanks.

badwolf1972, 2020-04-20:
Thank you for the reply first yes i am using it through USB and pi 4 has 2 USB 2 AND 2 USB 3 ports i have tried all of them with the same issue.... linux does not keep log files in the location you gave me logs under linux are normally in /var/logs i have looked in there and even done a search for cura.log i got no hits at all ???????

smartavionics, 2020-04-20:
[image]

badwolf1972, 2020-04-20:
here is the log thank you 
[cura.log](https://github.com/smartavionics/Cura/files/4503197/cura.log)

smartavionics, 2020-04-20:
Thanks for the file. Unfortunately, it doesn't show any messages related to the USB failing so I can't say what is going wrong. Personally, I don't use Cura to drive the printer directly, I use Octoprint. I have a Pi 2 and a Pi 3 both driving printers that use Marlin and have no problems at all. So maybe the easiest solution is to switch to using Octoprint rather than Cura to drive the printer?

badwolf1972, 2020-04-22:
hi there mark i tried octoprint like you said printed ok first time round however started getting timeouts and the printer would just stop mid print this is a snippet of the console log 

Recv: ok N129860 P0 B15
Send: N129861 G1 X30.293 Y102.813 E2377.78711*92
Recv: ok N129861 P0 B15
Send: N129862 G1 X30.543 Y101.849 E2377.82023*90
Recv: ok N129862 P0 B15
Send: N129863 G1 X30.547 Y99.688 E2377.89211*100
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N129864 M105*23
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N129865 M105*22
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N129866 M105*21
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N129867 M105*20
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N129868 M105*27
No response from printer after 6 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Printing" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor

i am not sure if its the pi4 or the firmware in the printer so i am going to print directly from sd card on the printer see if i have any drop outs then

eskeyaar, 2020-04-28:
Hi,
Sometimes I do have these kinds of issues when i run on different baudrate. So can you switch to different baudrates and try to print with Repetier host or Octoprint, try and see if that fixes the issue?