smartavionics/Cura #54: Last few windows builds have a cx.Freeze: Python error "Error in sys.excepthook" when starting

andymozza, 2020-05-07:
Hi Mark, I tried to install both the 1st and 5th builds this month from the dropbox but am getting a Python error on Windows (screenshot attached)

[image]

I just double checked with a main cura download and I don't get this with the official build which I just tested with to make sure - so was just wondering if you know what dll / pre-req this relates to so I can either cleanup, or install it?

Many thanks for input.

4.6.1 ( From Main Cura site does not have the issue )
1st May and 5th May builds from your site show the error and will not run ( exits out )

**Platform**
Windows 10

**Printer**
Doesn't get that far to be loading printers / extruders etc.

smartavionics, 2020-05-07:
Hi, I don't know what the issue is there. Can you locate the stderr.log file and see what's in there? Maybe also cura.log says something useful.

I do know of a problem if you have both 4.6.1 and my builds in use which is that 4.6.1 uses a higher version number on the config files than my builds currently do. I am making a new build now that uses the same version number and maybe that will fix the issue. I will let you know when that is available.

smartavionics, 2020-05-07:
There's a 20200507 build now available for Windows 64, please see if that one runs. Thanks.

andymozza, 2020-05-07:
No luck - I spent a little time this evening looking. The build doesn't change the behaviour and actually  the message pops up almost immediately after start attempt, so both stderr, stdout and cura.log in %localappdata%\roaming\cura are not getting anything written to them.

smartavionics, 2020-05-08:
OK, i'll dig into this and see what else changed for those releases.

smartavionics, 2020-05-08:
At the moment, I can't get it to fail. Unfortunately, I don't have any other Windows systems to install it on. Do you?

andymozza, 2020-05-08:
Hi Mark - thanks for looking into it - It is resolved for me and I wrote up what I did to help others if they have the same issue.
I backed up - then wiped out the old data from my %localappdata% for all the 'master' folders in local and roaming sub folders and also did an uninstall - removing profiles.

I noticed that the "program files\ultimaker cura master" was not empty - and there were a load of python files in there.
So likely problem was just left overs from previous bulids.

I think it's because I have been installing your builds for a while and no matter what the underlying cura version is, it all gets put into the 'master' folder - -whereas ultimaker obviously install new copy under the 'new' version number ( I have 4.0 and 4.6 for example).

So summary - full clean-up and re-install fixed it. I was able to copy back the profiles I backed up and all good.

smartavionics, 2020-05-08:
Hi, thanks for the update. Yes, it's not great that my releases install on top of each other. Not sure how to get around that unless I use a different installation directory for every release and have some code to migrate the settings from the previous master release.

Anyway, let's keep this issue open in case anyone else has the same problem or we want to talk some more about it.