smartavionics/Cura #55: Saving project no longer includes models

loneFunction, 2020-05-12:
**Application Version**
20200510

**Platform**
MacOS 10.15.1

**Printer**

**Steps to Reproduce**
Save project.

**Actual Results**
Project 3mf is saved, but much smaller filesize. No models included. When attempting to export just the models error occurs.

loneFunction, 2020-05-12:
[image]

loneFunction, 2020-05-12:
2020-05-12 15:53:27,141 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [151]: Writing to Local File /Users/cnicks/Personal/3D Printing/Projects/COVID-19-3dprints/printfarm/MP10_Mini/0.4 Nozzle/PLA/MODELS_PLA_04_1X_USCSW-v3L_MP10mini.3mf in binary mode
2020-05-12 15:53:27,159 - ERROR - [Thread-6] UM.Logger.logException [107]: Exception: Error writing zip file
2020-05-12 15:53:27,169 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [346]: Request [679ee9f831ec42729eadbd4203232e35] started
2020-05-12 15:53:27,172 - ERROR - [Thread-6] UM.Logger.logException [111]: Traceback (most recent call last):
2020-05-12 15:53:27,182 - ERROR - [Thread-6] UM.Logger.logException [111]:   File "/Applications/3D Printing/cura-mb-master-20200510.app/Contents/MacOS/plugins/plugins/3MFWriter/ThreeMFWriter.py", line 165, in write
2020-05-12 15:53:27,182 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-05-12 15:53:27,188 - ERROR - [Thread-6] UM.Logger.logException [111]:     savitar_node = self._convertUMNodeToSavitarNode(root_child, transformation_matrix)
2020-05-12 15:53:27,193 - ERROR - [Thread-6] UM.Logger.logException [111]:   File "/Applications/3D Printing/cura-mb-master-20200510.app/Contents/MacOS/plugins/plugins/3MFWriter/ThreeMFWriter.py", line 80, in _convertUMNodeToSavitarNode
2020-05-12 15:53:27,198 - ERROR - [Thread-6] UM.Logger.logException [111]:     savitar_node.setName(um_node.getName())
2020-05-12 15:53:27,205 - ERROR - [Thread-6] UM.Logger.logException [111]: AttributeError: 'SceneNode' object has no attribute 'setName'
2020-05-12 15:53:27,212 - DEBUG - [Thread-6] UM.FileHandler.WriteFileJob.run [80]: Writing file took 0.05737805366516113 seconds
2020-05-12 15:53:27,490 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [408]: Request [679ee9f831ec42729eadbd4203232e35] finished.
2020-05-12 15:53:27,494 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [317]: No more requests to process, stop
2020-05-12 15:53:27,495 - INFO - [MainThread] SliceInfoPlugin.SliceInfo._onRequestFinished [275]: SliceInfo sent successfully

smartavionics, 2020-05-13:
Sorry about that, I will investigate.

smartavionics, 2020-05-13:
I have fixed the problem (a library hadn't been updated) and made a new build, 20200513. Please give it a go.

loneFunction, 2020-05-13:
That fixed it. Much appreciated, thank you.

closed 2020-05-13 by loneFunction