smartavionics/Cura #56: Mesh Modifier for infill density cannot slice. Error with gradient infill

loneFunction, 2020-05-15:
**Application Version**

cura-mb-master-20200513

**Platform**

MacOS 10.15.1

**Printer**

Monoprice MP10 (Custom)

**Steps to Reproduce**

Place mesh as modifier for overlaps. Add Infill Density as setting and set to 0. Gradient infill is disabled.
[InfillMod_PLA_06_1X_mask-v3L_MP10.zip](https://github.com/smartavionics/Cura/files/4632852/InfillMod_PLA_06_1X_mask-v3L_MP10.zip)

**Actual Results**

Errors slicing

**Expected results**

Infill in areas of overlap would be removed.

**Additional Information**

smartavionics, 2020-05-15:
Hi, thanks for the report. Sorry about that, I can see what I have done and will fix today.

smartavionics, 2020-05-15:
Could you please try the 20200515 build.

loneFunction, 2020-05-15:
[image]
Thanks for getting on this. It is still unable to slice.

smartavionics, 2020-05-15:
Hmm, could you try modifying that setting and see if that message goes away? Thanks.

loneFunction, 2020-05-15:
[image]
Gradient infill is set to None. Infill Density on the modifier set to 0 still errors, but setting it to 1 slices successfully.

loneFunction, 2020-05-15:
[image]
Looks like when it does slice that it is still including some infill in areas where the overlap occurs.
I'm trying to remove all infill above the lower section to reduce travels. I wish Cura could print sections at a time instead of feature types. That way one isolated area could have infill, then walls printed and move on to the next island, instead of printing an infill line, then moving to another area and repeating. With poor quality printers/oozy nozzles that constant revisiting the same area reduces finish quality drastically. One of my biggest gripes with Cura.

smartavionics, 2020-05-15:
Something is not right here.  There must be some cached file that hasn't got replaced by the 20200515 release. Could you please try removing the contents of ~/Library/Application\ Support/cura/master/cache

loneFunction, 2020-05-15:
Good call. That did it. I'll make sure to include that process with every update.
Thank you again for your work on this.

closed 2020-05-15 by loneFunction

smartavionics, 2020-05-15:
Good, glad it's fixed, sorry for the hassle.