smartavionics/Cura #60: [4.6.1/20200525]Can't use 10key/numpad to input numbers.

earldbjr, 2020-05-27:
**Application Version**
4.6.1 / Cura-mb-master-x86_64-20200525
(Though it's been this way for a very long time)

**Platform**
kubuntu 20.04

**Printer**
Makes no difference

**Steps to Reproduce**

1) Start appimage
2) Select text field of any setting
3) Use 10key to enter new value

**Actual Results**

Camera changes angle corresponding to the key pressed. Happens whether text input box is active or not.  Happens with numlock on or off. Happens with Fn pressed or not.
**Expected results**

I expected the number pressed to be entered into the setting.
**Additional Information**

If this is by design then I'll deal with it, but it feels like a bug. Your builds are still far superior to the regular Cura releases. Thanks for doing what you do!

smartavionics, 2020-05-27:
OK, I'll look into it.

smartavionics, 2020-05-27:
I took a quick look and it seems that the numlock state isn't tracked at all in Qt or Cura. I guess it would be possible to do in Cura but given that I never use the numeric pad, I'm not really that bothered. Sorry.

closed 2020-05-27 by smartavionics

Sophist-UK, 2020-05-27:
I assume that @earldbjr has checked his numlock setting... - I have never, ever made that mistake myself, of course... ;-)

Otherwise it could be some sort of QT setting - with a fix needed in the Ultimaker Cura repo. It is on my To-Do list to look at how Escape and Alt-Space are handled by QT in Cura anyway, so when I do I will take a look at this too.

smartavionics, 2020-05-27:
It would be best if a fix went into the UM code base as this issue is not specific to my builds.

earldbjr, 2020-05-27:
That's interesting, because I can 10key properly in regular Cura builds.

smartavionics, 2020-05-27:
> That's interesting, because I can 10key properly in regular Cura builds.

In that case, it must be because my builds are using Qt 5.13.2 and (I think) UM are still using 5.10.

Sorry, I'm still not going to do anything about it.

earldbjr, 2020-05-27:
Thanks for the insight!

I used your Qt version as the basis for some searching and found mention that pressing shift while using the numpad works around the problem. It held true for your version of Cura as well. 

Thought I'd post it here in case someone else comes along.

smartavionics, 2020-05-27:
Thanks for that info, who would have guessed?