smartavionics/Cura #61: ".gcode" not appended to filename when saving gcode.

earldbjr, 2020-05-27:
**Application Version**

At least 20190823 through 20200525

**Platform**

Kubuntu 19.04-20.04

**Printer**

N/A

**Steps to Reproduce**

Slice Model
Click Save to File
Enter filename, Click Save or press Enter

**Actual Results**

Gcode is saved without filename extension, unless explicitly added by the user.

**Expected results**

.gcode is appended to the filename if not explicitly added by the user.

**Additional Information**

smartavionics, 2020-05-27:
It works for me on Ubuntu 16.04.6. What does the file dialog show? Mine looks like this when I click on save to file...

[image]

What is the default file type shown?

earldbjr, 2020-05-27:
Gcode, same as yours. 
Now if you remove ".gcode" from the filename in your screenshot and click save, will it still save as a .gcode?

For a real world example, if you're copy/pasting a name to use as a filename, normally you can just paste it in and .gcode gets re-appended, but as it currently works for me I have to highlight up to the .gcode and paste.

smartavionics, 2020-05-28:
> Now if you remove ".gcode" from the filename in your screenshot and click save, will it still save as a .gcode?

No, if you remove the suffix it doesn't append another one. I wouldn't expect it to.

Do the UM releases behave the same?

earldbjr, 2020-05-28:
Hmm they did when I was on Windows, but now that I'm on Linux it doesn't seem to be behaving the same way, so nevermind!

closed 2020-05-28 by earldbjr