smartavionics/Cura #62: Preferences/Printers/Machine settings - dialog boxes unreadable

Jens5395, 2020-06-05:
**Application Version**
Cura-mb-master-x86_64-20200525.AppImage

**Platform**
Linux

**Printer**
any

**Steps to Reproduce**
Go to Preferences Printers Machine setting and half the text in the text boxes is missing

**Actual Results**
I can't read what is in the entry box ... ie nozzle size

**Expected results**
would be nice to see the nozzle size  (and any other numerics on that page ... text is ok for some reason

**Additional Information**
If I remember correctly, this issue is not present in the official Cura release
This issue has been present for a long time 

[image]

smartavionics, 2020-06-05:
Hi, yes, it's a pain, isn't it. I see similar but not quite as bad.

[image]

If the UM releases don't show the same problem (and I'm not sure whether that's true or not) then it will be down to the fact that I am using a more recent version of Qt than UM do.

I'll investigate although this UI stuff is not really my area.

smartavionics, 2020-06-05:
Could you please try the 20200605 release?

Jens5395, 2020-06-05:
Replied via email ..... problem fixed!
Thank you very much!

closed 2020-06-05 by Jens5395

smartavionics, 2020-06-05:
That's good, I'll send the fix upstream so that they can put it in UM builds if they wish.