smartavionics/Cura #64: Compact purge tower collisions

Jens5395, 2020-06-30:
**Application Version**
Cura-mb-master-x86_64-20200605.AppImage,

**Platform**
Ubuntu 20.04

**Printer**

**Steps to Reproduce**
none needed,

**Actual Results**
Cura is causing nozzle to hit compact purge tower

**Expected results**
purge tower should still remain instead of being mowed over

**Additional Information**

I am doing a two material print, 188 layers, first 148 layers are the same material, with a compact purge tower.
The first 148 layers are of the same material and all goes well. Each layer purges and prints
When the second material comes into play, the hot end goes to the purge tower and promptly rips off the tower.

Here is the relevant section of g-code:

M566 X1200 Y1200
M566 X1800 Y1800
G1 F1800 E1618.01314
G0 F600 X94.191 Y106 Z15
G0 F12000 X62.372 Y125.729
;TIME_ELAPSED:4337.753071
;LAYER:148
G1 F600 Z15.5 ......... z hop was 0.5 mm, current print level was at 15 mm
G92 E0
G0 X100 Y100 Z10 F6000 .......... here the hot end is told to go to where the purge tower is
but the Z level is well below the height of the tower and promptly
collides with the tower

T0 ......... here the new hot end is selected followed by temperatures etc
G92 E0
M109 S205
M104 T1 S239.4
M106 S255
M104 T1 S240
G0 F12000 X62.372 Y125.729 Z15.5
M104 S195
M566 X900 Y900
;TYPE:SUPPORT-INTERFACE
G1 F600 Z15
G1 F1200 E6.15752
G1 F1600 X62.372 Y125.545 E6.16364
G1 X62.43 Y125.545 E6.16557
M566 X1800 Y1800

Comments are added to explain what is happening.

I can not figure out if there is anything in Cura that would cause it to drop the nozzle for the purge tower

.3mf file attached but renamed to a .txt extension

[linkloc round mount b.txt](https://github.com/smartavionics/Cura/files/4849435/linkloc.round.mount.b.txt)

smartavionics, 2020-06-30:
Could you please provide the project file? The file you have supplied only contains the models and not any settings. Do File -> Save and zip/rename the .3mf file, thanks.

Jens5395, 2020-06-30:
[linkloc round mount b second try.txt](https://github.com/smartavionics/Cura/files/4849632/linkloc.round.mount.b.second.try.txt)

Sorry about that .... didn't know there were different .3mf files

smartavionics, 2020-06-30:
Thanks for the project file, unfortunately, it isn't the project that produced the above gcode, the first 148 layers do not use the same material for all objects. Please provide the project file that produces the above problem. Thanks.

[image]

Jens5395, 2020-06-30:
OK, I do not know what is going on here any more. It must be past the witching hour and I must have turned into a pumpkin or something. In any which case, I find myself unable to provide a file for now.
More specifically, I set everything up and slice it and then save the .3mf file. Then, to make sure everything is right this time, I load that .3mf file into Cura but even though the parameters say to use a compact tower, when I slice it, it uses a full tower for some odd reason.
So, I haven't got a clue what is going on at this point so I will head to bed and try to sort this out tomorrow morning.
Sorry about that.
I restarted Cura and loaded the file again but at this time I have no idea what you will see on your end because I am afraid to re-load the file into Cura
[linkloc third try.txt](https://github.com/smartavionics/Cura/files/4849818/linkloc.third.try.txt)

So after I uploaded the file I tried to load it into Cura and sure enough it came across with a full purge tower even though my settings are set to the compact purge tower. I tried deselecting compact purge tower, slicing it and then re-selecting compact purge tower and re-slicing but it still thinks it is doing a full purge tower.

smartavionics, 2020-06-30:
OK, no problem.

Jens5395, 2020-06-30:
Oh man .... rough day .... somehow or other I ended up loading one model directly over top of the existing model. One was set up for one extruder, the other for the second extruder so yes, the prime tower ended up the same as a full tower.
Heading to bed now to curl up and die of embarrassment .....

Jens5395, 2020-06-30:
Ok, the last file I posted yesterday was in fact the file that shows the problem --- linkloc third try.txt (which downloads as linkloc.third.try.txt and needs to be renamed with a .3mf extension)
Please let me know if that works out for you.

smartavionics, 2020-06-30:
It's you, it's you!!!!

[image]

Jens5395, 2020-06-30:
OMG   <facepalm> ........ WTF was I thinking?????  Well actually I know what I was thinking but this was set up about a year ago and obviously had a MAJOR unintended consequence!
Unbelievable .....

Thank you so much for figuring this out for me, I literally spent days trying to find the issue!

smartavionics, 2020-06-30:
You're welcome (you can get some sleep now!)

Jens5395, 2020-06-30:
At the risk of exposing my lack of knowledge even further .....
The same model as above, while printing the section where only one material is used, Cura alternates the fan output between layers as if it was printing alternate material layers.
The cooling fan is set up as 50% for one material and 100% for the second material. During the period where only one material is used, Cura will run at 50% for two layers then switch to 100% for the next two layers and then back to 50%.
This would be proper operation if I was using both materials but makes no sense when only one material is being printed.
This is not a serious issue but thought I would mention it. Hopefully it is not another operator problem .....

smartavionics, 2020-07-01:
Hi, when I look at the gcode on that last example, the fan goes to 100% at layer 221 (or thereabouts) and stays at that speed to the end. At lower layers the fan alternates between 50 and 100% as the materials change. So isn't that right?

Jens5395, 2020-07-01:
Assuming I didn't mess it up again, the purge tower is a compact tower and there is only one material being used for the first 3/4 or so of the print before the material alternates between the two extruders. What should be happening (in my world anyway) is that while one material is used, there should only be one fan speed. When the second material is used and the material alternates, the fan speed should alternate.
I should mention that I only have a single part cooling fan instead of one fan on each hot end.

smartavionics, 2020-07-01:
In the project you sent me, there's two materials being used for the first 220 or so layers and one material after that...

[image]

Jens5395, 2020-07-01:
That is the project that was sent by mistake. Look at the project here : https://github.com/smartavionics/Cura/issues/64#issuecomment-651574812

smartavionics, 2020-07-01:
That is the third try project...

[image]

Jens5395, 2020-07-01:
I loaded the project from here and I get this (after slicing for the second extruder)

[image]

Jens5395, 2020-07-01:
Doing a couple more tests, it seems that the saved project doesn't actually carry all the setting information with it. 
I cleared the 'use compact prime tower' setting in Cura and exited. I then started Cura again and loaded the project again. The loaded project should have had the flag to 'use a compact prime tower' set .... however, it did not. From this I assume that not all settings are saved (and reloaded) with a project file.

smartavionics, 2020-07-01:
I can see that there are some problems with the compact prime tower and this project. I'll look into it and get back to you when I can.

smartavionics, 2020-07-02:
I think I have fixed this, uploading a 20200702 release now but my broadband is very slow so it will take 30 mins.

I don't have a printer I can test this on so I would be grateful if you can thoroughly test and let me know if there are further issues.

Jens5395, 2020-07-02:
Will do ... tomorrow ....
Can't believe it wasn't an old fart error again :)
Thanks!

Jens5395, 2020-07-02:
The fan is behaving exactly as it should!

Thank You !

smartavionics, 2020-07-03:
That's good. I'll close this issue now, please open another if there are further problems or you have any suggestions for the compact prime tower.

closed 2020-07-03 by smartavionics

Jens5395, 2020-07-03:
Thanks again!